polarofficial/polar-ble-sdk #379: Wrong "Disk Full" Error in Offlinerecording mode after device goes into energy saving mode

merschel, 2023-06-15:
**Platform on which you observed the bug:** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

get disk full error although disk is nearly empty after energy saving mode  in normal mode. 

**How to Reproduce**

1. clear all data from the device
2. go into normal mode
3. start offline recording (acc+ppg)
4. close app
5. do not wear the device on the skin, so the device will turn into energy saving mode after 10 minutes
6. start the app and restart the device
7. connect the device 
8. try to restart the offline recording
9. see in log: 

`iosBleSdkTestApp[61143:3862904] failed to start offline recording ppg. Reason: controlPointRequestFailed(errorCode: 14, description: "Disk full")`

after deleting the few data the offline recording can be started again

**Expected behavior**

offline recording starts

jimmyzumthurm, 2023-06-26:
Hi @merschel , I apologize for the delayed answer. I've been trying to reproduce by following the steps you provided but without success for now, but this definitely sounds like a bug. Does this happen only after device turns off automatically after 10 minutes of not being worn ? In the other ticket I saw that you noticed the recordings were automatically stopped, even when device didn't shut himself down. As you said, both issues could be related. Here are all the events that could stop all ongoing offline recordings at once without requesting a stop from Polar SDK :

- Device got connected to charger
- Long button press -> device shutdown
- Device has not been able to calculated HR for 10 minutes -> device shutdown
- Device reaches <= 3% of battery level -> device shutdown
- Free memory goes under 300 KB (see right under)

About memory : A fresh device should have around ~14 MB of free memory. Note that a minimum of 2 MB of free space is required to be able to start a new recording, or to enable triggered recordings. As I said, all ongoing recordings automatically stops when 300 KB of free space limit is reached. This is to have enough margin for users to still be able to do Polar exercise recordings (swim or other indoor). 

It seems that you already brought the getDiskSpace() API into use, which is great. Could you tell me what total space and free space are reported by the API ?

An other thing : if you haven't done it already, maybe you could try to perform a factory reset to the device ? This will erase the volume used by the file system, so it might help. Here is how to do it : https://support.polar.com/e_manuals/verity-sense/polar-verity-sense-user-manual-english/resetting-the-sensor.htm 

If you have more details related to this issue or other issues, please do not hesitate to report them, it helps us quite a bit.

merschel, 2023-06-29:
Hi @jimmyzumthurm,

thank you for answering the question.

First, I tested the problem with two other devices and I could not reproduce the problem. But with the original device I can reproduce the problem every time. I tested your cases in the list, and with the problematic device, if I test long enough, I get the "disk full" error. But on the other devices it works as expected. 

I appreciate you pointing out the existence of the getDiskSpace() API. I was not aware of it because I was working with the 5.10 tagged commit. Now that I have used this API, I believe we are closer to identifying the problem.

On the devices where the problem does not occur, I ended up with 14 MB of free space after deleting all the data.

However, on the problematic device, I only have 3 MB of free space after deleting everything.

The listOfflineRecordings() API doesn't list any more data, so it seems that the device retains some data that I can't delete. So maybe the listOfflineRecordings() API has a problem.

This would explain this problem and the other problem in normal mode (https://github.com/polarofficial/polar-ble-sdk/issues/355). 

Unfortunately, I can't explain how I got the device into this state.  

So far, I have not done a factory reset. I think if I do, the problem will be solved, but maybe for now it is best to keep the device in the buggy state, and if you want, I can run more tests or send you some log files or something else to identify the bug, since I can't explain how I got into this state. Please let me know.

merschel, 2023-07-04:
Hi @jimmyzumthurm,

After further testing, I found a way to reproduce the problem:

1) Delete all data from the device
2) The getDiskSpace() API should show about 14 MB of free space. 
3) Start an offline measurement in normal mode with PPG and ACC 
4) Wear the device for several hours so that the memory is full. 
5) List all packets using the listOfflineRecordings() API. There should be at least 2 (one PPG and one ACC).
6) Enter the delete command for all packages at once. So don't delete all packets one after the other but all at once. Here, the device usually loses the connection (it seems as if it switches itself off and on again).
7) After the connection is re-established, check if there are still packets on the device. If yes, delete them.
8) If all packets are deleted, use the getDiskSpace() API to get the free space, you should see only a few MB left.

jimmyzumthurm, 2023-09-08:
Hi @merschel ,
Thank you for the detailed report and steps to reproduce. With errorlogs you provided, we were able to find a problem. Device was crashing during the big file erasing, leaving file system in a dirty state were only part of the file was being erased. So the file was not listed anymore, but some data was left over, and this is why it returns that disk is full. Fix is ready for next firmware version, so I'm closing this ticket now.

closed 2023-09-08 by jimmyzumthurm