polarofficial/polar-ble-sdk #382: What are possible ways to preserve Polar battery during a continuous connection to a device?

talbigelbinah, 2023-06-28:
***Platform your question concerns:*** 
- [x] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
What is the best way to prolong the battery life of Polar Verity Sense over a continuous use? I tested and saw that when keeping the Polar connected to a device and Online Streaming PPG with 135Hz, the Polar battery lasts 18 hours.
If I use a lower sampling rate, for example 28Hz, would the battery last longer? Or if the Polar is just connected and not sampling would the battery last longer?

Or is there some other way to preserve the battery while the Polar is still connected to the mobile device?

andYgnite, 2023-07-19:
Hi @talbigelbinah,

can you please details more what is your final application?

Are you setting the sensor in SDK mode or are you in normal mode?
Indeed streaming raw PPG data reduces battery life.

andYgnite, 2023-07-20:
Hi @talbigelbinah 

In normal mode, there is a 10-minute timer that automatically turns off the sensor if it cannot detect a valid heart rate.
This timer is however disabled when in SDK mode.

Turning off streaming will indeed enhance the battery life. 
You can then run some test and make an average to see how much extra battery life you can get.

talbigelbinah, 2023-07-24:
Thanks for the response @andYgnite! We'll give it a try.

closed 2023-09-08 by jimmyzumthurm