polarofficial/polar-ble-sdk #383: deviceDisconnected not called after PolarH10 disconnected

AndrezVazz, 2023-06-29:
**Platform on which you observed the bug:** 
- [x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
After enabling EcgStreaming, if the PolarH10 devices gets disconnected from the band, the deviceDisconnected callback never gets called, and the values of the ecg signals never stops. When this EcgStreaming function is not used, deviceDisconnected is called (even with some delay, around 10 seconds).
This happens on the **AndroidBleSdkTestApp**.

**How to Reproduce**
Enable the startEcgStreaming feature -> Connect the PolarH10 -> Check the values on log -> Remove PolarH10 -> Check that deviceDisconnected is never called. 

**Expected behavior**
After removing the device, deviceDisconnected should be called and the EcgStreaming should be stoped, or throw an error

kitkatbonsai, 2023-07-20:
Hello @AndrezVazz, could you please tell us in which Android model are you having this problem, please? Thanks in advance.

AndrezVazz, 2023-07-25:
Hello @kitkatbonsai ! I'm having trouble in a Xiaomi Mi 11 Lite 5G running Android 13 with mask MIUI Global 14.0.5

andrezzvazz, 2023-08-21:
Hi @kitkatbonsai ! Do you have any news on thi topic? Thanks in advance.

closed 2025-07-03 by rkangast