polarofficial/polar-ble-sdk #387: Multiple EcgData at one timestamp.

latownjo, 2023-07-24:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I have dumped all EcgData and combined them into one table. If you sort this data by the timestamp of the EcgSample, the frame rate will be sometimes incredible high. Further I analysed the timestamps of the EcgData package and those timestamps jump back and forward. So it seems, that you have two packages at once.

**How to Reproduce**
Dump all EcgData and analyse the timestamps.

**Expected behavior**
Continuous ecg data record with roughly same sample rate.

**Screenshots and logs**
First 5h of a 24h test recording. Rest of the data has no issues.
[image]
This are the EcgSample timestamp differences (ns, log scale). They fluctuate between 0 and 7685000 ns (130.1Hz). Sometimes the frame rate is constant, but increases periodically. Also there are some gaps with data loss(positive spikes).
[image]
This are the timestamps (in ms) of the EcgData (all zero values were removed for better visualisation), sorted by the absolute timestamps of the EcgSample. So, if everything is right the EcgData is dumped each 500ms, but sometimes we have 2 EcgData for the same EcgSample timestamp.

latownjo, 2023-07-25:
IT seems that the SDK were connected to multiple devices.
[image]

aburashima, 2023-10-13:
Hi latownjo,

Could you please tell me how I can launch ECG data stream on Polar H10 device?
I tried to use API call startEcgStreaming(...) on having received bleSdkFeatureReady for .ecg feature, but got error "controlPointRequestFailed(errorCode: 5, description: \"Invalid parameter\")".
What "settings" values do you use for calling startEcgStreaming(...)?

Thank you in advance!

latownjo, 2023-10-16:
Hi @aburashima,

I used the normal API: [requestStreamSettings](https://github.com/polarofficial/polar-ble-sdk/blob/c0f3648fee9db7fb3065f9c28b90af87b03fea3e/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarOnlineStreamingApi.swift#L35) and called [maxSettings](https://github.com/polarofficial/polar-ble-sdk/blob/c0f3648fee9db7fb3065f9c28b90af87b03fea3e/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/model/PolarSensorSetting.swift#L57) on that class.

Hope the info helps.

aburashima, 2023-10-16:
> Hi @aburashima,
>
> I used the normal API: [...]

Yes, it helped! Thank you so much!