polarofficial/polar-ble-sdk #389: BLE error "Peer removed pairing information" not handled properly (lost BT pairing keys?)

orestesgaolin, 2023-07-28:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
We don't know the reason but sometimes Verity Sense looses BLE pairing keys. We received multiple reports from our users that when this happens then the only way to resolve it is to forget the device via system settings. See more about this particular error [here](https://developer.apple.com/forums/thread/132488).

After first occurrence of the `Peer removed pairing information` the SDK tries to continuously reconnect with the device.

Logs are shown below.

Seems like the [didFailToConnect](https://github.com/polarofficial/polar-ble-sdk/blob/43d4a38f240a000ec7498eb2a023765d086a523e/sources/iOS/ios-communications/Sources/iOSCommunications/ble/endpoints/corebluetooth/central/CBDeviceListenerImpl.swift#L202-L212) is not exposed anywhere else beside the log and then passed to `handleDisconnected` which in turn tries to connect with the device again, repeating the cycle.

**How to Reproduce**

We don't really have a way of reproducing this issue. If you have any ideas we'd be happy to validate them.

**Expected behavior**

I would expect the SDK to inform the consumer about the failure, maybe along with the failure reason. I think that adding failure flag to `deviceDisconnected` or exposing another method in the `PolarBleApiObserver` protocol would be sufficient.

**Screenshots and logs**

<details><summary>Logs from iOS</summary>
<p>

```
2023-07-28T07:42:05.449Z BFS: Starting service for device ABCDEFGH. Restored: true
2023-07-28T07:42:05.605Z PC: Setting heart rate listener
2023-07-28T07:42:05.605Z BFS: Received device battery update {"value": -1, "timestamp": "1690530125605", "deviceId": "" }
2023-07-28T07:42:05.606Z Polar SDK log: [BLE] willRestoreState called with dict: ["kCBRestoredScanOptions": {
    kCBScanOptionAllowDuplicates = 1;
}, "kCBRestoredPeripherals": <__NSArrayM 0x28221cc30>(
<CBPeripheral: 0x281721ad0, identifier = 7264988F-7707-52B0-3ED8-0DD2B46A1D07, name = Polar Sense ABCDEFGH, mtu = 23, state = connecting>
)
, "kCBRestoredScanServices": <__NSArrayM 0x2822113e0>(
Heart Rate,
FEEE
)
]
2023-07-28T07:42:05.606Z Polar SDK log: [BLE] willRestoreState found devices: Optional(<__NSArrayM 0x28221cc30>(
<CBPeripheral: 0x281721ad0, identifier = 7264988F-7707-52B0-3ED8-0DD2B46A1D07, name = Polar Sense ABCDEFGH, mtu = 23, state = connecting>
)
)
2023-07-28T07:42:05.606Z Polar SDK log: [BLE] willRestoreState found device: <CBPeripheral: 0x281721ad0, identifier = 7264988F-7707-52B0-3ED8-0DD2B46A1D07, name = Polar Sense ABCDEFGH, mtu = 23, state = connecting>
2023-07-28T07:42:05.608Z PC: Device battery updated: {"value": -1, "timestamp": "1690530125605", "deviceId": "" }
2023-07-28T07:42:05.608Z PC: Setting device synchronization listener
2023-07-28T07:42:05.608Z PC: Setting device version listener
2023-07-28T07:42:05.608Z BFS: Received device version update {"deviceId": "", "softwareVersion": "" }
2023-07-28T07:42:05.608Z PC: Device version updated: {"deviceId": "", "softwareVersion": "" }
2023-07-28T07:42:05.609Z PC: Setting device contact listener
2023-07-28T07:42:05.609Z BFS: Received device contact update {"deviceId": "ABCDEFGH", "timestamp": "1690467570242" }
2023-07-28T07:42:05.609Z PC: Device contact updated: {"deviceId": "ABCDEFGH", "timestamp": "1690467570242" }
2023-07-28T07:42:05.609Z PC: Connecting to device ABCDEFGH
2023-07-28T07:42:05.609Z BFS: Received connection update {"connection": 0, "deviceId": "ABCDEFGH", "name": "Polar Sense ABCDEFGH" }
2023-07-28T07:42:05.609Z PC: Device connection updated: {"connection": 0, "deviceId": "ABCDEFGH", "name": "Polar Sense ABCDEFGH" }
2023-07-28T07:42:05.611Z Polar SDK log: [BLE] commandState state:IDLE action: CLIENT_START_SCAN
2023-07-28T07:42:05.611Z Polar SDK log: [BLE] commandState state:IDLE action: EXIT
2023-07-28T07:42:05.611Z Polar SDK log: [BLE] commandState state:SCANNING action: ENTRY
2023-07-28T07:42:05.611Z Polar SDK log: [BLE] start scan services: nil
[... 38 lines cut ...]
2023-07-28T08:28:08.760Z Polar SDK log: [BLE] default session state update from:  sessionOpenPark  to:  sessionOpening
2023-07-28T08:28:08.761Z Polar SDK log: [BLE] commandState state:SCANNING action: CLIENT_REMOVED
2023-07-28T08:28:08.761Z Polar SDK log: [BLE] commandState state:SCANNING action: EXIT
2023-07-28T08:28:08.762Z Polar SDK log: [BLE] commandState state:IDLE action: ENTRY
2023-07-28T08:28:09.156Z Polar SDK log: [BLE] didFailToConnect:  <CBPeripheral: 0x281725450, identifier = 7264988F-7707-52B0-3ED8-0DD2B46A1D07, name = Polar Sense ABCDEFGH, mtu = 0, state = disconnected> error:  Peer removed pairing information
2023-07-28T08:28:09.163Z Polar SDK log: [BLE] default session state update from:  sessionOpening  to:  sessionOpenPark
2023-07-28T08:28:09.163Z Polar SDK log: [BLE] commandState state:IDLE action: ENTRY
2023-07-28T08:28:09.163Z Polar SDK log: [BLE] commandState state:IDLE action: EXIT
2023-07-28T08:28:09.163Z Polar SDK log: [BLE] commandState state:SCANNING action: ENTRY
2023-07-28T08:28:09.163Z Polar SDK log: [BLE] start scan services: nil
2023-07-28T08:28:09.228Z Polar SDK log: [BLE] default session state update from:  sessionOpenPark  to:  sessionOpening
2023-07-28T08:28:09.228Z Polar SDK log: [BLE] commandState state:SCANNING action: CLIENT_REMOVED
2023-07-28T08:28:09.228Z Polar SDK log: [BLE] commandState state:SCANNING action: EXIT
2023-07-28T08:28:09.228Z Polar SDK log: [BLE] commandState state:IDLE action: ENTRY
2023-07-28T08:28:09.608Z Polar SDK log: [BLE] didFailToConnect:  <CBPeripheral: 0x281725450, identifier = 7264988F-7707-52B0-3ED8-0DD2B46A1D07, name = Polar Sense ABCDEFGH, mtu = 0, state = disconnected> error:  Peer removed pairing information
2023-07-28T08:28:09.609Z Polar SDK log: [BLE] default session state update from:  sessionOpening  to:  sessionOpenPark
2023-07-28T08:28:09.609Z Polar SDK log: [BLE] commandState state:IDLE action: ENTRY
2023-07-28T08:28:09.609Z Polar SDK log: [BLE] commandState state:IDLE action: EXIT
2023-07-28T08:28:09.609Z Polar SDK log: [BLE] commandState state:SCANNING action: ENTRY
2023-07-28T08:28:09.610Z Polar SDK log: [BLE] start scan services: nil
2023-07-28T08:28:09.681Z Polar SDK log: [BLE] default session state update from:  sessionOpenPark  to:  sessionOpening
```

</p>
</details>

samulimaa, 2023-07-31:
Hello,

Explanation and possible fix:
iOS applications are not permitted to remove Bluetooth pairings, so user has to remove the pairing manually from the settings.

To detect when this is happening, you could try monitoring the device connection state with [PolarBleApiObserver](https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApiObservers.swift#L7) and attempt to detect when the device gets stuck in the device connecting state. Then, for example, a popup could be shown to the user, guiding them to remove the pairing from the Bluetooth settings and re-pair the device. For usage of PolarBleApiObserver, please see example implementation from iOS example app: https://github.com/polarofficial/polar-ble-sdk/blob/master/examples/example-ios/iosBleSdkTestApp/iosBleSdkTestApp/PolarBleSdkManager.swift#L1164

orestesgaolin, 2023-07-31:
We are already using `PolarBleApiObserver` in our implementation and it reports that the device is connected, the `deviceConnected` is being called in an infinite loop. The `deviceDisconnected` is not being called.

samulimaa, 2023-08-11:
The SDK indeed had a bug where `PolarBleApiObserver.deviceDisconnected()` was not called in the case of 'Peer removed pairing information.' This has now been fixed in SDK version 5.2.0. Also `deviceDisconnected()` now has a `pairingError` parameter, which indicates that the device was disconnected due to an error in pairing. Your application can then detect this and guide the end user to remove pairing from the system settings.

I was able to reproduce this issue with Verity Sense by pairing the sensor with a phone and then performing a factory reset on the sensor. This should cause the pairing keys to be lost. Here's how to perform a factory reset: https://support.polar.com/en/how-can-i-reset-my-sensor

orestesgaolin, 2023-08-11:
Thank you so much for addressing the issue, we appreciate it a lot!

Wondering if you ever encountered this issue when the pairing keys would be lost "spontaneously" with no explicit factory reset?

samulimaa, 2023-08-14:
No problem.

No, strange, it doesn't seem to be widely reported issue.

closed 2023-09-08 by jimmyzumthurm

orestesgaolin, 2023-09-28:
Hey there, I noticed you added new ability to do the factory reset remotely in 5.4.0. Is it necessary to pair with the device first to perform the factory reset? We were wondering if this might be possible workaround for pairing/connection issues that we're experiencing sometimes, but if we cannot connect to the device in the first place (even via Polar Flow), then this will kot work anyway, right?

samulimaa, 2023-09-28:
Hey, it's unfortunately not possible to do factory reset if device is not connected.