polarofficial/polar-ble-sdk #390: How to prevent the sensor from shutting down automatically

ardi2150, 2023-08-08:
***Platform your question concerns:*** 
- [ ] Android 
- [*] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [*] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
When using the sensor we noticed that if a sensor broadcasts "0" for heart rate for about 10 minutes, the sensor automatically shuts down. Is there a way to prevent this from happening? Is there anything in the sensor SDK mode that can be manipulated to stop the sensor from shutting down?

andYgnite, 2023-08-14:
Hi @ardi2150 ,

Verity Sense has a timer that automatically turns off the sensor after 10 minutes  if any activity is detected (for instance 0 bpm).
We are planning to release a new version with the timer disabled when in SDK mode.

ardi2150, 2023-08-15:
Thank you @andYgnite. Any ETA on the new firmware version? In the new version, would we be able to keep the sensor from shutting off, without the SDK mode?

andYgnite, 2023-08-16:
Hi @ardi2150,

no ETA yet. Verity Sense in normal mode will keep the 10 minutes timer.

jimmyzumthurm, 2023-09-08:
Hi @ardi2150 , there is a work-around where you need to wear the device when entering SDK mode, then device will not shutdown himself automatically. Basically when HR is 0 (algorithm was unable to calculate HR, usually if device is not worn), there is a timer that is started and shutdown device after 10 minutes. As Andrea, in "normal" mode this behavior will be kept, as it can prevent battery drain when user put his device in a drawer but forgets to press the button to shutdown device. 

In 2.1.0 there is a bug where this timer is not stopped when entering SDK mode, but if timer is not running when SDK mode is entered, device will not shutdown. Note also than online streaming of raw data should turn off this timer as well.

closed 2023-09-08 by jimmyzumthurm