polarofficial/polar-ble-sdk #391: ECG data from PolarH10 using Python

paoloart, 2023-08-16:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hello,

I'm working on a project where I aim to extract real-time ECG data from the Polar H10 heart rate monitor using a Raspberry Pi and BLE (Bluetooth Low Energy). While I understand that the Polar H10 primarily measures heart rate, I've come across mentions that it might also provide raw ECG data or related metrics.

Could someone help me with the following aspects:

Setting up a BLE connection between the Raspberry Pi and the Polar H10 specifically for accessing ECG data.
Recommended Python libraries or SDKs that would facilitate the extraction of ECG data from the Polar H10.
Guidance on discovering and connecting to the relevant services and characteristics provided by the Polar H10 that pertain to ECG data.
Steps or code examples for streaming and decoding the ECG data in Python.
Any help, tutorials, or pointers related to this would be immensely valuable. Thank you in advance for your insights!

Best regards,

NgPhuong21163, 2023-08-17:
Hello,

Your project is very interesting! Here are the aspects that you can consider to perform ECG data extraction from Polar H10 heart rate monitor using Raspberry Pi and BLE.

Set up BLE Connection:
To establish a BLE connection between Raspberry Pi and Polar H10, you need to use a BLE library in Python. A popular option is the bluepy library. You need to learn how to scan and connect to your Polar H10 device via BLE. Once the connection is successful, you can access the services and features of the device to obtain ECG data.

Python library or SDK:
As mentioned, bluepy is a popular BLE library. You can also try using the pyGatt library to interact with the BLE device. These libraries will help you facilitate the extraction of ECG data from Polar H10.

Discover Polar H10 Services and Features:
Once connected, you need to explore the services and features that Polar H10 offers regarding ECG data. Normally, ECG data is transmitted via BLE characteristics. You need to specify the characteristic that holds the ECG data and other related characteristics such as data format, update frequency, etc.

Streaming and Decoding ECG Data in Python:
After defining the feature that contains the ECG data, you can read the data from the feature through the BLE connection. ECG data is usually encoded as integers or decimal values. You need to see the Polar H10 documentation for specific data formats. You can then decode and process the ECG data as required by your project.

Be sure to check the documentation and instructions from Polar about using BLE with your Polar H10 device, as the BLE protocol and how data is accessed may vary by device version.

Good luck in your project!

Best regards,

paoloart, 2023-08-25:
Hi @NgPhuong21163 

I sincerely thank you for taking the time to respond to my question and for providing such a detailed overview of the process for extracting ECG data from the Polar H10 using Raspberry Pi and BLE. Your insights have been incredibly helpful!

I've followed your advice, and I've managed to connect to the device and access the heart rate and RR interval data. However, I'm encountering an obstacle with the raw ECG data.

After several attempts, it seems that the specific characteristic for ECG data might be this handle: 0x0039, uuid: fb005c52-02e7-f387-1cad-8acd2d8df0c8. But if it is this one, it appears to be locked, and I can't read the data.

Do you happen to know if the ECG data is locked and if it's accessible only through a specific SDK? I've looked through the documentation and online resources but haven't found clear information about this.

Any further assistance or guidance would be truly valuable for my project. Thank you once again for your support and expertise.

Best regards,

erikboto, 2023-09-08:
The ECG data is not locked in any way, but I think Polars focus is on providing SDKs for some platforms and therefore the documentation on how to read it without those SDKs is pretty sparse. 

I would start at looking at a PDF that was removed a while back, but it's still in the git history: https://github.com/polarofficial/polar-ble-sdk/blob/v2.2.3-beta1/technical_documentation/Polar_Measurement_Data_Specification.pdf

It's not very detailed, but it shows some examples of the flow to enable a stream of a certain type etc which I believe it still valid today. I think that the data formats used might have developed over time though, and you will probably need to have a look at either the iOS or Android SDK code for details on how to interpret the ECG data, and perhaps also for which settings you need to send to enable a stream. I think they have some compressed formats now that needs a bit more handling before you get the actual data.

So it can definitely be done, but it requires some effort.

Chakib-Temal, 2023-09-20:
Hello 

I read your question and find your project interesting. 

I'm working on a Flutter (Mobile) project, and I'd like to do the same as you. I'm already using a BLE plugin to communicate with lots of other sensors.

I'd like to add this one to my list, but is there any clear documentation on how to read the data (order and reception)? 

did you find more documentation ?

Chris-Spooner999, 2023-12-03:
Did you come across this - https://github.com/fsmeraldi/bleakheart/ ? I have tested it and ECG acquisition works on Linux. Only problem is there is sometimes a 30 delay in ECG data coming though. I am currently trying to fix this problem that I am also having in my C++ implementation of a Polar H10 Bluetooth LE interface. Both this Python implementation and my C++ code are having the same problem.

closed 2025-07-03 by rkangast