polarofficial/polar-ble-sdk #393: polar-ble-sdk-5.2.0.pom doesn't exist on jitpack.io due to build failure

alansley, 2023-08-22:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this bug

**Describe the bug**
Trying to obtain `polar-ble-sdk-5.2.0.pom` returns "Build failled - see log at jitpack.io" - 5.1.0 is still available though.

**How to Reproduce**
Go to the following URL and see the error message:
https://jitpack.io/com/github/polarofficial/polar-ble-sdk/5.2.0/polar-ble-sdk-5.2.0.pom

Also going to the following shows that the files are not there:
https://jitpack.io/com/github/polarofficial/polar-ble-sdk/5.2.0/

While if we step back a version to 5.1.0 the files are listed as existing:
https://jitpack.io/com/github/polarofficial/polar-ble-sdk/5.1.0/

**Expected behavior**
polar-ble-sdk-5.2.0.pom builds & exists on jitpack.io

**Screenshots and logs**
N/A

Rexios80, 2023-09-06:
Same with version 5.3.0. This is pretty urgent please fix asap.

alansley, 2023-09-06:
Come on Polar, this is pretty important.

I know your repo has instructions to build the .aar, but the jitpack.io POM is how builds should work via gradle: I want X, here is X.

Please fix your 5.2.0 / 5.3.0 jitpack builds for the sake of 3rd party devs using your hardware - you know it makes sense =D

samulimaa, 2023-09-07:
Hello, This is now fixed in 5.3.0 version.
https://jitpack.io/com/github/polarofficial/polar-ble-sdk/5.3.0/

closed 2023-09-08 by samulimaa