polarofficial/polar-ble-sdk #394: Feature Request - Flag indicating recording state on PolarOfflineRecordingEntry

Preventicus-Sebastian, 2023-08-28:
### The Problem

When listing all on the device stored offline recording entries (via. _listOfflineRecordings()_), there is currently **no information whether a recording is currently in progress or already finished**. Moreover it is possible to download offline recordings which are currently in progress. 

This makes downloading offline recordings hard. When blindly downloading all recordings returned by _listOfflineRecordings()_ you might download data twice. You could develop a workaround, by sorting the offline recording entries by date, take the last element for a specific recording type in the list, and check if recording is in progress for that type. But this is rather complicated. 

### Possible Solution

It would be really helpful to have access to a property on PolarOfflineRecordingEntry, which indicates if a recording is completed or currently in progress. This could for example be a simple Boolean flag like _isFinished_ or _isInProgress_.

jimmyzumthurm, 2023-09-07:
Hello @Preventicus-Sebastian ,
Thank you for your input and experience feedback using the offline recording feature. 

> 
 Moreover it is possible to download offline recordings which are currently in progress.

For next firmware release, we had in mind to not list anymore the recordings which are ongoing to prevent potential issues, so that the user has to properly stop the recording. This can have drawback where it is not possible to know the current recording file size before stopping the recording.

jimmyzumthurm, 2023-09-07:
Hi @Preventicus-Sebastian  ,
Thanks the for the feedback and suggestion, I open a discussion about that internally. 

> Moreover it is possible to download offline recordings which are currently in progress.

We are planning to remove that functionality in the next firmware release, we found out that this can lead to unwanted bugs / side effects, meaning that a recording will have to be stopped before it can be transferred back to mobile. This would somehow prevent to download same data twice when downloading all at once. Please let me know if you see it as problematic that it is no longer possible to fetch an active recordings or if you have other suggestions.

Preventicus-Sebastian, 2023-09-07:
Hi @jimmyzumthurm 

Thank you for your response!

Personally I would prefer some flag on _PolarOfflineRecordingEntry_ like described above. This would enable you to decide yourself if a _inProgress_ recording should be downloaded or not. (Since you need the entry object anyway to start a download)

But besides that removing the possibility to download _inProgress_ recordings completely would not limit us in any way. In fact it would eliminate our issues with duplicated downloads. So this solution would also be nice for us!