polarofficial/polar-ble-sdk #395: 1.1.5 -> 2.1.0 issues

zarpli, 2023-08-30:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

hi,

I use microcontrollers with BLE to get data from the Verity Sense sensors,

everything was fine with firmware 1.1.5, but now with version 2.1.0 I have the following problems:

**1.** The sensor does not send data unless you add a delay of at least 100ms between starting the SDK mode and the PPG request:

```
// Enable SDK Mode
if(pRemoteCharacteristic->canWrite()) pRemoteCharacteristic->writeValue(SDK_packet, sizeof(SDK_packet), true);

delay(100); // WTF?!

// PPG request
if(pRemoteCharacteristic->canWrite()) pRemoteCharacteristic->writeValue(PPG_packet, sizeof(PPG_packet), true);
```

Is there a way to know if the sensor is _busy_?

**2.** With version 1.1.5 it was possible to put the sensor in SDK mode and get PPG data even when the sensor was connected to the USB charger.

Is there a way to do the same with version 2.1.0 ?

https://youtu.be/KwZjZR4B7JA

jimmyzumthurm, 2023-09-07:
Hi @zarpli , thank you very much for the detailed bug report. 

About issue 1, this is definitely an unwanted regression if that delay was not needed in 1.1.5 and is now needed in 2.1.0. Currently there is no possible way to know that device is busy. What happens is that when entering SDK mode, device needs to shutdown every of its component (algorithms, sensors) so everything is ready to be reconfigured by Polar SDK user with custom settings. It might be possible that 2.1.0 added some delay as we added new features, and if everything happens too fast, it is still in the process of shutting down PPG sensor when PPG measurement start is sent by SDK app, but this sound like a bug and it should work regardless. We will investigate that and see if that can be fixed for next release. For 2.1.0, I'll advise to keep that delay unfortunately. 

About issue 2 : I tried to reproduce on my side, and it seems the issue is where SDK mode is started when device is already connected to charger. Work around is to start SDK mode while device is not charging, and plugging it afterwards. In this case it is possible to stream PPG data. This will be fixed for next release. Also thank you for giving example of using the streaming feature with a charger connected, we were not aware of people using device in such a way, it seems like a nice project you have here !

zarpli, 2023-09-08:
hi @jimmyzumthurm,

although developers like us (museums, DIY, artists, etc.) do not represent an important market, we greatly appreciate that you promote and invest in this support!

ale.

closed 2023-10-12 by samulimaa