polarofficial/polar-ble-sdk #396: Polar H10 Internal Recording doesn't work for long measurement - 24 hrs

carldatu, 2023-09-13:
**Platform on which you observed the bug:** 
- [X] Android 
- [X] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I tried to start internal recording with Polar H10 device for 24 hrs. After finish the training,  I could not get the recording sample data from Polar H10 device.

**How to Reproduce**
Start internal recording for long time (ex. 24 hrs). Finish recording. 
When try to get recording data from that exercise, app doesn't work.

**Expected behavior**
Polar H10 Manual says max internal recording duration is 30 hrs. so for 24 hrs internal recording, it should work properly.

**Screenshots and logs**
Here is log when get the recording data from 24 hrs measurement.
So basically, To get recording sample data, app call the `fetchExercise` api for Android and it raise the below error.
`com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpOperationTimeout: Air packet was not received in required timeline`

carldatu, 2023-12-11:
Thanks, I hope to fix it ASAP.

infomus, 2024-02-26:
We're experiencing the same issue on Android A8 Tab when calling startRecording. Is there any update on this yet?

orestesgaolin, 2024-03-12:
Wondering if there's any update on `Air packet was not received in required timeline` issues?

smyffanon, 2024-05-11:
> Wondering if there's any update on Air packet was not received in required timeline issues?
for me the frequency of the issue went way down when my Moto g8 power broke and I went to a google 6a.

It's been happening now once every 3 to six months instead of 2x per month.

palmqvisti, 2025-02-04:
Polar internal task FLOW-62389.

itsymbal, 2025-05-08:
Same issue. I have the same (H10, latest FW 3.3.1; using the latest Android SDK 6.2.0).
My own app and the Polar Android sample app are experiencing the same issue. 
"Air packet was not received in required timeline"

Attaching the complete log. Polar API uses tags "PolarBleApi" in the attached log.

[Polar Air Packet Not Received.txt](https://github.com/user-attachments/files/20107386/Polar.Air.Packet.Not.Received.txt)

itsymbal, 2025-05-08:
> FLOW-62389.

Any update on this issue? We're losing a lot of data because of this bug.

palmqvisti, 2025-05-09:
@itsymbal, unfortunately no progress yet with this issue.