polarofficial/polar-ble-sdk #397: implement polar H10 with ionic framework

abdelkarimmamen, 2023-09-15:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] ionic
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I want to use "ionic framework" to build a cross-platform application to connect with polar H10 but I didn't find in [polar-ble-sdk](https://github.com/polarofficial/polar-ble-sdk/tree/master) any documentation about it.
Can someone provide me with documentation or an example?

samulimaa, 2023-10-12:
Hello, unfortunately only native Android/iOS platforms are officially supported by Polar.

closed 2023-10-12 by samulimaa

abdelkarimmamen, 2023-10-12:
> There is no specific sdk for hybrid mobile framework such as ionic. You need to develop the ionic sdk by using native ios and android sdk that Polar provide. To do it, I suggest you to use bridge concept for ionic.

can you provide me with docs that do similar things to understand what you mean, thank you in advance