polarofficial/polar-ble-sdk #399: Feature Request: Add the device type to the PolarDeviceInfo object

talbigelbinah, 2023-09-19:
***Platform your question concerns:*** 
- [x] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this question

**Description:**

Less of a question, more a feature request of adding to the PolarDeviceInfo object a 'type' or a 'deviceType' field with the type of the of the Polar device (Whether it's Verity Sense, H10, etc.).

The use-case is we would like to only 'show' certain device types (for example, only Polar Verity Sense) when searching for devices using the searchForDevice() API, and filter ones we don't support.

samulimaa, 2023-10-02:
Hello, have you tried using [PolarDeviceInfo.name](https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarDeviceInfo.kt#L26) field?

In the example app, Polar device name is logged when scanning devices: https://github.com/polarofficial/polar-ble-sdk/blob/master/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt#L257

The output in the app would be something like: `polar device found id: 8C4CAD2D address: A0:9E:1A:8C:4C:AD rssi: -46 name: Polar Sense 8C4CAD2D isConnectable: true`

Note that "Polar Verity Sense" name is just "Polar Sense" with device ID.

talbigelbinah, 2023-10-10:
Thanks. We'll use the .name field.

But is it still possible to maybe add a 'type' field in a future release (as either a declared const or enum)?

The reasons are mostly:

1. If the name scheme ever changes, it would break anything that relies on the current scheme.
2. It would save the need to parse or rely on string.contains plays.

samulimaa, 2023-10-12:
Yes I created internal ticket about adding the "type" field in future releases.

closed 2023-10-12 by samulimaa