polarofficial/polar-ble-sdk #405: PPI Stream Failed: Reason com.polar.sdk.api.errors.PolarNotificationNotEnabled

RitikaNigam-ux, 2023-10-16:
***Platform your question concerns:*** 
- [* ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [* ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I have integrated the HR and PPI stream API in my unity game android app. I am able to record HR (Bpm) while running the game but PPI streams fail due to com.polar.sdk.api.errors.PolarNotificationNotEnabled. I wanted to know how to solve this error and enable PPI stream.

I have attached the log which shows the error and also code snippet which I am streaming PPI

<img width="1512" alt="Screenshot 2023-10-16 at 6 14 26 PM" src="https://github.com/polarofficial/polar-ble-sdk/assets/7318090
[image]
3/60d96120-2272-4dab-8c53-5d635bfe3b7f">

RitikaNigam-ux, 2023-10-16:
[image]

chrisgblr, 2023-11-09:
I have the same issue on my Pixel 8 Pro with Android 14. Could be related to Android 14.
My Verity Sense is paired successfully in my bluetooth settings.
I get the error when i try to start an offline recording on Verity Sense:
`PlatformException(com.polar.sdk.api.errors.PolarNotificationNotEnabled`

Disconnecting and reconnecting with the Verity Sense through the SDK doesn't fix the problem.
The only solution i found was disabling bluetooth and enabling it again. After that the error is gone for a certain amount of time.

jimmyzumthurm, 2024-03-06:
hi @RitikaNigam-ux ,
Sorry for the late answer, but looking at your logs it seems that you try to enable the PPI streaming feature right after the connection is established but before the Polar Measurement Data Service characteristics notification are subscribed. 

I recommend that you try to enable PPI streaming using `bleSdkFeatureReady` callback

closed 2024-03-06 by jimmyzumthurm