polarofficial/polar-ble-sdk #408: "libmagtsync.so" not found

adojnov, 2023-10-22:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

Hello, 
I am building an App but when I try running the App, the app crashes and I get the following message: Can't load library: dlopen failed: library "libmagtsync.so" not found. Does anyone know where I could get this library?

Thank you kindly, 
AD

poovarasan-1998, 2024-01-23:
same error for me that is when we launch the project on android 13

samulimaa, 2024-03-13:
Hello, where you able to solve this? I have not encountered this issue before, but you could make sure you have updated Android SDK Build-Tools from: Android Studio -> SDK Manager -> Tools -> SDK Tools.