polarofficial/polar-ble-sdk #409: How can I get all recorded offline sessions by SDK? Polar Verity Sense

mayander30, 2023-10-27:
***Platform your question concerns:*** 
- [ ] Android 
- [ X] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I am running the IOS example project, but in the offline tab, it does no show anything.

How can I get all recorded offline sessions?

mayander30, 2023-10-27:
Only Polar Flow app can get the offline workouts?

jimmyzumthurm, 2023-10-30:
Hi @mayander30 ,

There seems to be confusion about offline recording of raw (ACC, PPG, MAG, GYRO) or computed (HR,PPI)  data from the SDK and internal workout recording with Verity Sense, see [here](https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/OfflineRecordingExplained.md).

You can sync and manage your workouts from the SDK using `listExercises`, `fetchExercise `and `removeExercise `APIs. I believe this is not part of the example project.

closed 2023-10-30 by jimmyzumthurm

mayander30, 2023-10-30:
I can't find listExercises for IOS SDK..