polarofficial/polar-ble-sdk #414: Delay in ECG and ACC transmission from H10  using Python

dara65, 2023-11-07:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [X] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I am using python (and Bleak) to collect, ECG, ACC and RR data from Polar H10. I noticed that there's a delay between starting notification and the start of data transmission (ECG and ACC) from the device. The delay is about 25-30 seconds after starting the notification and receiving data. Any idea why there's such delay and what the problem is?

Here's my device info:

> Hardware revision:  00760690.03  

> Software revision:  3.2.0  

> Firmware revision:  5.0.0 

I should note that I receive RR interval data with no delay, so I have a significant mismatch between my ECG and RR interval timestamp data.
```
    RR_UUID = "00002A37-0000-1000-8000-00805F9B34FB"
    await client.start_notify(RR_UUID, heart_rate_data_handler)
    await client.write_gatt_char(PMD_CONTROL, ECG_START_STREAM)
    await client.write_gatt_char(PMD_CONTROL, ACC_START_STREAM)
    await client.start_notify(PMD_DATA, ecg_data_conv)
    await client.start_notify(PMD_DATA, acc_data_conv)
```

abdelkarimmamen, 2023-11-07:
good morning,
I want to gather ECG data from Polar H10 and I'm using Ionic can you tell how to do it in Python so I can inspire from you? 
because I can only read HR and RR intervals.

jimmyzumthurm, 2023-11-07:
Hello @dara65 , 
To reduce the data payload transmitted over the air, the data arrives in batches that contain several samples which are compressed into a buffer that is as big as the ATT_MTU  of the connection allows. The data will be sent only after this buffer is full, thus the delay. The data is not completely real-time and this is why we provided timestamps with nanoseconds resolution to each sample so you can make fusion with other data.

dara65, 2023-11-07:
@abdelkarimmamen check out this repo 
[pypolartest](https://github.com/stuartlynne/pypolartest).

dara65, 2023-11-07:
@jimmyzumthurm would you tell me please what you mean by :
> "you can make fusion with other data."

the RR intervals have no timestamp data and ACC is also transmitted with the same ATT_MTU size.

jimmyzumthurm, 2023-11-07:
@dara65 My answer wasn't very clear. All **raw** data we transmit with our proprietary service (ACC, ECG, MAG, GYRO, PPG, ....) have the same timestamp basis (nanoseconds with 01.01.2000 00:00:00 UTC, assuming time was properly set using setLocalTime API beforehand) so I was meaning that especially. My comment about ATT_MTU was applying to those measurements and not R-R intervals. 

R-R intervals are sent over BLE Heart Rate service which does not specify a timestamp field unfortunately.

closed 2023-11-07 by jimmyzumthurm

dara65, 2023-11-07:
@jimmyzumthurm Thank you. Is there any way/API to set the local time in python?

jimmyzumthurm, 2023-11-07:
@dara65 You would have to implement that functionality, as we do not provide official support for other platforms than iOS and Android. You can look at the implementation of this API in Kotlin/Swift code and port it to python, that should be possible.