polarofficial/polar-ble-sdk #418: frequency sampling rate in Polar OH1

aminhm2002, 2023-11-22:
***Platform your question concerns:*** 
- [ ] Android 

***Device:***
- [ ] Polar OH1

**Description:**
Hi
I am working with the polar OH1.
I see the same problem with the sampling rate. I used your experiences but the quality of signal is really weak! it is 8 samples per cycle.
the picture of the signal is in below:
[image]

I do appreciate it if let me know how many peaks you receive in each second? and how many points you have in each PPG cycle?

Best Regards
Amin

aminhm2002, 2023-11-23:
[JOikarinen](https://github.com/JOikarinen) ca you help me?

andYgnite, 2025-07-14:
@aminhm2002 

What is the x axis? Is it seconds?
In any case, that signal looks like noise only without any PPG.

OH1 sampling rate is 135Hz and the number of peaks depends on HR. If you have heart rate of 60 bpm, you'll get 1 peak per second

closed 2025-07-14 by andYgnite