polarofficial/polar-ble-sdk #427: Broadcast advertisement and Manufacturer data format

jcaude, 2023-12-16:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I want to use the HR data broadcast by a Polar H10. I am able to retrieve the advertisement data and the Manufacturer data of 4 bytes. It seems that the last two bytes are HR values (u8) ? But what about the first two bytes ? At first I thought it would be some RR signal but after converting to u16 it doesn’t make sens. Any tips ? Am I correct for the last two ?

vimmers2020, 2024-05-22:
Hi, Did you figure it out? I  have the same question and can't find the answer anywhere. And yes, I think you're right about the 2 bytes with HR data.

jcaude, 2024-05-22:
Still not found any documentation about it. But, yes it's R-R data in little endian format (2x u8 -> u16).

vimmers2020, 2024-05-22:
Super, thanks!

jcelerier, 2024-08-30:
Some other Polar devices such as H10 provide heart rate variance (HRV), maybe the first two bytes are this ? As described in this SO question: https://stackoverflow.com/a/52935617/1495627

vimmers2020, 2024-08-30:
Hi,

Thank you!
Once I proceed with this hobby project (when the darker evenings come along
again) I’ll dive in it and will use this info for sure.

Best,
Luc