polarofficial/polar-ble-sdk #432: iOS example not working

itssagunkarki, 2024-01-24:
**Platform on which you observed the bug:** 
- [ ] Android 
- [ - ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Describe the bug**
Description of your observation

`pftp_request.pb`

```{swift}
#if swift(>=4.2)

extension Communications_PbPFtpOperation.Command {
  // Support synthesized by the compiler.
}

#endif  // swift(>=4.2)

```

**How to Reproduce**
The GitHub actions are throwing some errors, and the build has failed.
When I try to build the example app, it shows an error somewhere in the file I posted above.

itssagunkarki, 2024-01-25:
I found that we need to install the older versions of SDK and upgrade it to the newer version, not just the newest one.

closed 2024-01-25 by itssagunkarki

YohayBar, 2024-04-16:
Had the same issue, 
based on the answer above here how it was solved for me:

1. downgrade to version 5.4.0 in PodFile

```
target 'iosBleSdkTestApp' do
    pod 'PolarBleSdk', '5.4.0'
end
```
1. run 
`pod install
`
which set an older SwiftProtobuf
`-> Installing SwiftProtobuf 1.26.0 (was 1.23.0)
`
1. set back to  5.5.0

```
target 'iosBleSdkTestApp' do
    pod 'PolarBleSdk', '5.5.0'
end
```
1. run` pod install ` again