polarofficial/polar-ble-sdk #437: Offline recording support

timonkot131, 2024-02-06:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Can I get offline records and start offline recording on OH1? I have read the docs and see what this might not be possible for OH1. But on the store page is written what it have 4mb of memory. Does it mean what band can record and hold data offline and it can be fetched?

jimmyzumthurm, 2024-02-07:
Hi @timonkot131 

Indeed OH1 has 4 MB of internal memory, but it doesn't support the offline recording feature as Verity Sense does, as this was introduced only 1 year ago through a new firmware update. Unfortunately there is no plan to update OH1 in the near future to offer this feature.

closed 2024-02-07 by jimmyzumthurm

maximilian-V, 2024-02-21:
Hello @jimmyzumthurm, does this mean that there is no way to receive the offline data from the OH1?

jimmyzumthurm, 2024-02-21:
@maximilian-V What do you mean exactly by offline data ? Offline recording is this : https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/OfflineRecordingExplained.md Maybe you are talking about internal trainings (swimming or other indoor) that you can start with device and then sync to Flow ?