polarofficial/polar-ble-sdk #438: PPG signal/sdk mode

jacobostapenko, 2024-02-06:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I am trying to stream PPG, PPI, ACC from verity sense but I am only getting the PPI signal. Anyone know why this could be happening? 

I also tried to enable SDK mode through python, using this as guidance:
"to enable the SDK mode you send command [0x02, 0x09] where the 0x02 = Request Measurement start and 0x09 = SDK Mode
to disable the SDK mode you send command [0x03, 0x09] where the 0x03 = Request Measurement stop and 0x09 = SDK Mode"

but I get a Bleak Error saying that action isn't allowed. is there another way to enable sdk mode?

jimmyzumthurm, 2024-02-07:
Hi @jacobostapenko 

For PPG and ACC you would need to pass settings alongside as part of the start command. For accelerometer sampling rate, range, number of channels and resolution. For PPG the same except the range. PPI has no settings so maybe that is why command is going through but not the others. 

The command sent to enable / disable SDK mode seems correct, but note if you are using Verity Sense 2.1.0 version or later, you need to first stop all measurements before you can enter SDK mode. Also note that PPI is not available in SDK mode as SDK mode doesn't have any Polar algorithms running.  

_"if the online stream or offline recording is currently running on the device, the SDK Mode cannot be changed. Attempting to change the SDK Mode in will result in an "INVALID STATE" error."_ : https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkModeExplained.md

abbjetmus, 2024-11-27:
If I don't use SDK Mode does it mean we can store PPI offline as well in normal mode, I don't understand here?

[https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkModeExplained.md](https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkModeExplained.md)

abbjetmus, 2024-11-27:
I'm new to the Polar SDK and I was not certain that if one doesn't use SDK mode that PPI would still be available offline which my use case require. But you confirmed that it is.

jimmyzumthurm, 2024-11-27:
@abbjetmus Yes it is available offline from version 2.1.0. SDK Mode is a special mode that is only useful when : 

1) You are not interested in any computed data from the device -> HR / PPI, but only in raw data provided by the different embedded sensors. 
2) You are interested to configure the sensors (ACC, MAG, GYRO, PPG) in a special way, for example high sampling rates.

We will improve the documentation very soon about this to clarify.

closed 2024-11-27 by jimmyzumthurm