polarofficial/polar-ble-sdk #440: `ERROR_INVALID_OP_CODE` when calling `isSDKModeEnabled`

Rexios80, 2024-02-12:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Calling `isSDKModeEnabled` gives an `ERROR_INVALID_OP_CODE` error

**How to Reproduce**
Call `isSDKModeEnabled` after receiving the SDK Mode feature ready callback

**Expected behavior**
No error

jimmyzumthurm, 2024-02-28:
Hi @Rexios80 ,
Which version of Verity Sense do you have in use ? It is a bit surprising that it gives error, but I believe this command was added in software release 2.1.0. Could you have older version than that ?

Rexios80, 2024-03-01:
Yep that's most definitely the issue. It was on version 1.1.5. I had to use the desktop app to update because the mobile app said there was no update available.

closed 2024-03-01 by Rexios80