polarofficial/polar-ble-sdk #442: responseError(errorCode: 202) when trying to listOfflineRecordings with Polar OH1

maximilian-V, 2024-02-21:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hello, 

I am currently trying to use listOfflineRecordings in an Expo Module:

`AsyncFunction("getOfflineEntries") { (deviceId: String, promise: Promise) in
            polarBleSdkManager.api.listOfflineRecordings(deviceId)
                .observe(on: MainScheduler.instance)
                .subscribe{ e in
                    switch e {
                    case .next(let entry):
                        self.offlineEntries.append(entry)
                    case .error(let err):
                        promise.resolve("Offline recording listing error: \(err)")
                    case .completed:
                        promise.resolve(self.offlineEntries)
                    }
                }.disposed(by: disposeBag)
        }`

Unfortunately I am getting the following error from my promise in the case .completed: responseError(errorCode: 202) 

Does someone know how I can fix this?

jimmyzumthurm, 2024-02-21:
Hello @maximilian-V ,
Unfortunately OH1 doesn't support the offline recording feature, only Verity Sense does currently. There is no plan to update OH1 to provide that functionality. The main reason is that internal memory of the device is very small and will get full very fast when recording sensor raw data.

maximilian-V, 2024-02-21:
Hello @jimmyzumthurm, does this mean I can only use the internal memory with the official polar applications such as Polar Beat?

jimmyzumthurm, 2024-02-21:
@maximilian-V It seems the API you want to use here is `listExercises `and not `listOfflineRecordings`. You should be able to sync training sessions like Beat or Flow does using that.

closed 2024-02-21 by jimmyzumthurm