polarofficial/polar-ble-sdk #443: Questions on ACC data format

giulianosrebuli, 2024-02-23:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hi, i am retrieving accelerometer data from my Verity Sense. I am using the PMD Specification for 3rd party document for data interpretation.

As by the example shown there, i am writing the following command to the PMD control characteristic:

`[0x02, 0x02, 0x00, 0x01, 0x34, 0x00, 0x01, 0x01, 0x10, 0x00, 0x02, 0x01, 0x08, 0x00, 0x04, 0x01, 0x03]`

The response i get after writing the command is this:
`[F0, 02, 02, 00, 00]`

And after that, the data stream starts, here are some samples:

227 bytes:
[02 00 41 6F C4 12 45 93 0A 81 EE 02 1F FE 82 FF 08 0A 10 DA 20 D6 A3 0B D9 CB E7 D6 F6 C4 E6 F1 E1 F3 E0 15 1A FE 28 1B F8 37 2C 25 20 13 19 11 0A 2F ED 4F C0 7E F4 E9 BB 6E 80 0D C8 2F B0 3E 14 D1 AF B0 7F 16 C4 CC 20 87 34 03 54 12 8B FF 8D 00 C1 42 FB E6 D3 DF B5 E1 91 53 2F BE F9 ED 43 B0 3E FD DB 9B 90 3D F5 03 60 5F 7B FA D9 DF AE 00 0B 24 64 70 40 12 CC CF BF 04 E4 C5 33 B0 38 08 41 80 EF C2 12 EC 17 81 46 FD FE 5B B0 01 F3 D1 D7 6F 81 FF CD 7B 0F C2 01 36 60 7F 43 D3 5F 3F 3F 78 CB 06 48 4E FF 00 0E FC 82 05 51 AB 55 E3 4A 2D 1D 0C 5F BD E8 58 73 4D BD EB A0 67 51 02 0C C4 70 41 CA 26 7F A0 02 89 19 FE 3F 5F 74 E7 22 44 20 3F 15 66 B0 52 4D 05 5C A4 1F BE E9 68 00]

229 bytes:
[02 94 04 51 07 13 45 93 0A 81 81 03 26 02 A1 02 08 01 1B 6C 27 0A 28 DE 0F 40 B8 0E 0D 84 80 00 F3 33 48 30 43 F0 63 77 8C B5 ED 8B A3 6B B1 87 0C FB CD AC C1 57 84 9D C4 0E 0F 5C 92 C6 18 84 E8 70 00 F7 F9 AB 2F FA 04 F6 83 40 80 F9 60 08 72 FA 22 37 98 02 85 E9 2D 70 CE 7D F9 02 B0 0D BF EE 8E 98 5E 07 E7 FE 07 91 3C F7 6C 7C 8F FB 07 F3 7F 0F BE FD E9 DF AF C3 0B 57 1C 6E FE 03 1D AC 5E C2 F9 2B 08 C2 BD 0D CB 37 A2 C8 F0 2A 88 B2 76 0B 37 44 C0 80 ED D1 D7 40 13 D8 34 70 0E 3F 0E 57 14 5E C5 19 D5 9F EF 45 09 08 12 11 F3 CC 1E 48 05 24 E0 D1 90 26 5E 48 EB E7 0C 0E 2F 0E 11 FD 14 12 25 F9 FB 01 E0 C2 09 E6 DE 35 C6 D3 49 DA A0 20 CC B0 2C C8 D3 35 D9 D6 1D E9 D3 25 CD CE 10]

226 bytes:
[02 B8 DD 51 4B 13 45 93 0A 81 55 02 49 FE BE 01 08 01 EA AE 1C 0A 14 F0 DF CF 42 F9 DC 3F 50 3F F9 E4 2F EF 3B 04 BC 53 5E 80 FB C4 4B E0 BF 06 31 EC C4 F2 11 BF 26 14 C4 FA 2E F4 7F C6 20 F2 BF 81 4B EF 39 48 C2 FB 0C 46 30 9F C3 00 8D B3 F0 81 E2 43 48 41 77 13 4D A0 5E 81 0D C4 87 40 05 FB 08 1E FD 02 43 BB 9F 51 CD BE 55 FD FE 32 07 11 1A 04 F9 F8 FC DD F7 00 16 08 08 04 04 F7 F1 0B F0 F6 18 F1 E2 15 EA FE 18 F4 FB 07 01 FB F7 FC 23 06 04 FB F2 E5 D9 DA 01 0C DA 29 40 E3 23 37 BA 1F 13 BA 08 10 D4 E8 31 14 EC 3B 61 F0 42 62 03 E1 26 EE 12 2C 00 1C 33 FA 2B 3A 0A 09 0A 48 31 40 FE D9 BF 30 02 0F CF 8F 9F BC F9 FE C3 EF 3B 01 DC A3 3E 3F F9 7B 27 60 BD EE DA 14 32 37]

228 bytes:
[02 E2 C1 DC 8A 13 45 93 0A 81 32 03 F3 00 47 00 0A 38 02 B0 FF 3E FD F4 A7 AF BE FD 07 CC 2F FB 07 D1 F7 DF C1 FE 1E 78 61 FF 1B 63 90 F0 CB 26 F8 67 80 01 03 44 24 20 80 0D F3 FF AF 42 00 DB D7 7F C1 00 FA 67 2F 00 FC EA 5F B0 01 FC 0E D4 D0 81 00 FB 8B 50 BF FD 16 40 C0 41 0B 02 60 90 81 FA B3 A7 3E 41 E9 AF 23 DF F2 E2 D7 6F 9C 7A FC 28 FB 9E 7C CC C8 1F 50 B7 05 E9 DF 2F 04 FC B1 2F 11 FE EA 10 84 1F BB 0A B2 4F 6E C0 F9 A8 E7 40 3A EB 8D 14 8E 78 24 C3 B3 FD C7 FB FC CF 70 82 21 FA B7 31 4E E9 6B 28 63 BA 1B B1 60 BE C5 17 93 8F A1 46 F2 7C 6C E2 FB 10 68 E4 3E 01 07 DC 6B 4F 7C 01 D4 3B 8F 03 04 1E D0 20 3F 0A 09 54 6E 7B 14 AB 97 CD C3 DC 5F 57 50 F5 D1 46 4C 4D AB 1B]

228 bytes:
[02 38 56 F9 CF 13 45 93 0A 81 07 00 84 FB D4 02 08 03 D4 A6 25 FE CD F6 1C 45 F3 0A 14 34 F4 61 C0 0C 55 38 D0 43 1B F3 43 D1 C8 F6 2D EC D1 FC 0F 1D 84 EF 43 15 DA 5F 41 45 F8 55 F8 A0 7B 15 30 78 AE 86 0F 96 F7 20 47 ED 53 B4 E1 7F F8 85 B8 C0 C2 F9 23 0C A0 7B FD DE 77 CF 00 02 12 20 B0 00 09 F5 DB 3F FD F5 08 1E C3 B4 11 BE 9E 23 C3 8A 18 C2 A2 30 C7 F0 4D ED EE 43 E1 D2 06 08 F0 0D FF F7 0E E5 0C 20 D1 06 26 E6 BD 07 D4 99 ED C8 AE 00 EC F2 0D 07 FF 1B FD F8 F8 EF 07 25 EC CF 37 02 1D 34 2F 67 06 48 6D C6 51 7F AB 3D 48 E4 04 13 0A 06 00 EF 0E 16 E2 08 F3 EA F4 01 EE D3 B4 16 0A 08 BF 4B 6E 86 F7 11 18 A2 FF 09 4D 10 E1 45 E9 73 F4 31 F7 27 CD 4C 9E C3 23 EB 03 00 03 FD]

228 bytes:
[02 E2 2B 15 15 14 45 93 0A 81 28 03 48 00 CB 01 08 0B F3 CC B8 F0 E0 EF 09 19 01 0D 21 15 09 F4 E9 12 0A E5 1B 2C FF 09 23 00 06 FA FF FA FB 21 08 FF 2C 0A 1E 10 F8 E0 02 FD 11 8C 60 BD EB 0F 84 BE B7 FA 96 DF FC BD F5 6B A3 9F 3E ED 08 94 1F BD 09 F3 87 EF C1 FC CB 1F 70 FF FC 24 CC 0F 3F 03 E3 6F EF 80 F6 C9 DF 5F 3F F5 E1 DF 4F 3B 05 DD CF 3F 84 03 7F F8 D0 45 30 DB 37 11 4D F1 5F 84 31 80 FE 18 E4 A0 FE F7 29 00 00 81 02 13 58 10 3F 0A 54 C4 CF 82 12 D4 8B E0 02 F2 0B 0C 90 7D 02 04 A8 0F 08 14 01 12 0C 02 17 0B EE E8 06 E5 B6 13 F0 D6 FD 1B 0C D5 FC 2F F2 16 14 F4 0C EF 1D FB EE 0D F4 E9 F8 E3 D6 FD CF C6 04 F5 D2 08 1C 15 EF 30 61 B7 31 4E A6 2B 2F A7 24 0B C4 09 14 EB]

The document says that the first byte is the PMD type (0x02 for accelerometer), bytes 01-08 are for timestamp and byte 09 for Frame Type, but at the byte 09, all i receive is 0x81 = 129, which is a RFU value, and i cant get to interpret neither as a simple 0-2 frame type nor as a 128 Delta Frame.

Also, I am not so sure what the Delta Frame is, and would apreciate if anyone could clarify it to me.

Am i doing something wrong here?

Thanks in advance!

jimmyzumthurm, 2024-02-26:
Hi @giulianosrebuli ,
The document you are using as a reference is probably pretty old and might not contain everything. This frame type is frame type 1 with delta compression which is indicated by a bit mask with 0x80. You would need to decompress these samples (examples on how to do are in SDK source code).

giulianosrebuli, 2024-02-26:
Hi @jimmyzumthurm, thanks for pointing it out to me.
I'll give it a try now, but it should solve my problems.

Thanks again

giulianosrebuli, 2024-02-27:
For future questioners, it seems to have solved my problems. Here is a sample of the way i decoded my package and it worked well for me:

for this data:
[02 2E 57 8A F1 37 32 94 0A 81 AE FF C2 FF E6 03 06 **3E 7D B2 13 F9 E1 23 3C D1** FB BE 1F 00 45 10 10 BC EF 07 82 51 00 79 FF F3 40 2F F8 83 20 F8 41 20 E8 81 60 14 3C E0 0F F7 FF 2F 3D 71 F7 B8 20 20 05 B0 F7 C1 EE EF 09 31 F0 F5 D1 37 B7 E0 2B 4A FC E8 89 BA 13 44 C0 0B 01 B1 FB 7F 3F 00 06 00 11 C0 CE F7 F6 4E 0C C3 DF 13 7F 9E 03 43 C0 17 BF B0 17 C5 E0 F3 C3 1E 0C 01 90 1B C2 F0 FF 3E E0 FB BA 10 20 00 B0 FB 3E F1 07 FC 70 0C C1 FD F3 42 41 18 7A 9F 03 FE DF 07 88 0F 04 14 E4 23 2C 4B 53 B2 DC 12 0E 1C 63 D2 EB 30 42 11 EB 2B 31 30 D1 0D 12 50 BD FF D3 E3 C0 D1 06 10 00 22 0C F9 1F FC 06 A0 E3 3F 52 FC 7E E0 0F 39 D1 27 7F D0 03 C0 2F 08 81 D0 EB BF 61 00 FB C0 17 7B 30 00]

* byte 0: 02 - PMD Type (ACC);
* bytes 1-8: 2E 57 8A F1 37 32 94 0A - timestamp (didn't really matter for me, though);
* byte 9: 81 (129) - is as @jimmyzumthurm pointed out, my data will be delta compressed, and the values will be referenced as 16-bit data;
* byte 10-15: AE FF C2 FF E6 03  - 48-bit x,y,z data as reference for the posterior data
--> x = 0xFFAE = -82 milli-g
--> y = 0xFFC2 = -62 milli-g
--> z = 0x03E6 = 998 milli-g
* byte 16: 06 - size of each delta-data, in bits, which means that every 6 bits is a different sample
* byte 17-end: delta data for 6 bits, in x,y,z, order. For this sample, i extracted the following:

```
   3E       7D       B2       13       F9       E1       23       3C       D1
00111110 01111101 10110010 00010011 11111001 11100001 00100011 00111100 11010001

```
and splitting every 6 bits:
```
001111 | 10 0111 | 1101 10 | 110010 | 000100 | 11 1111 | 1001 11 | 100001 | 001000 | 11 0011 | 1100 11 | 010001
```

which would result in the following sequence:

```
0        | 1        | 2        | 3
x  y  z  | x  y  z  | x  y  z  | x  y  z
0F 27 36 | 32 04 3F | 27 21 08 | 33 33 11
```
I interpreted these as signed 6-bit data, so the values are:
15 -25 -10 | -14 4 -1 | -25 -31 8 | -13 -13 17

and as it is a delta compressed frame, the absolute values are:

0:
--> x = -82 + 15 = -67 mg
--> y = -62 + (-25) = -87 mg
--> z = 998 + (-10) = 988 mg

1: 
--> x = - 67 - 14 = -81
--> y = - 87 + 4 = -83
--> z = 988 - 1 = 987

2:
--> x = -81 - 25 = -106
--> y =  -83 - 31 = - 114
--> z = 987 + 8 = 995

3:
--> x = -106 - 13 = -119
--> y = - 114 - 13 = -127
--> z = 995 + 17 = 1012

so on and so forth.

After analyzing a bunch of data, this made sense for me.
If anyone had the same question, i hope this explanation aid you in some way.