polarofficial/polar-ble-sdk #445: Polar Verity Sense Optical heart rate sensor

rextea, 2024-03-04:
***Platform your question concerns:*** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
When receiving PPG readings from the device, I have 4 different channels of readings:
`[
      {
        "timeStamp": 536614827214192000,
        "channelSamples": [380537, 435635, 480333, 184225]
      },
      {
        "timeStamp": 536614827221600000,
        "channelSamples": [380580, 435675, 480483, 184183]
      },
      {
        "timeStamp": 536614827229007000,
        "channelSamples": [380624, 435696, 480645, 184177]
      },...
]`
I'm trying to understand what are the difference and purpose of each one of them? and which one of them should I use to analyze and extract metrics such as P P intervals, heart rate, HrV metrics, etc...

Thanks in advance :)

jimmyzumthurm, 2024-03-04:
Hi @rextea ,
Verity Sense has 3 pairs of green LEDs used for PPG signal sampling, so 3 channels in total and the fourth one is ambient channel that is used for ambient noised cancellation by the HR/PPI algorithm in use in Verity Sense. I'm no algorithm expert and I can't really give deep guidance about handling this data, but you should be able to calculate the PP-intervals using a single of the 3 PPG channels as a starter.

rextea, 2024-03-17:
> Hi @rextea , Verity Sense has 3 pairs of green LEDs used for PPG signal sampling, so 3 channels in total and the fourth one is ambient channel that is used for ambient noise cancellation by the HR/PPI algorithm in use in Verity Sense. I'm no algorithm expert and I can't really give deep guidance about handling this data, but you should be able to calculate the PP-intervals using a single of the 3 PPG channels as a starter.

Understood. many thanks :)

closed 2024-03-17 by rextea