polarofficial/polar-ble-sdk #448: ignite 3 - getting the ppi through BLE-SDK

TinDogDev, 2024-03-17:
***Platform your question concerns:*** 
- [ X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Other
- [ ] Device is not relevant for this question

**Description:**
Hi,
As maintained at the Readme file, the Ignite 3 has a support for the BLE-SDK.
It supposed to provide PPI, HR and ACC.
When I used the sdk, and checked it using the ```getAvailableOnlineStreamDataTypes``` call, it is indeed supports the needed features.
The Issue is that i fail to get the PPI from the SDK.
I get the following message:
`PPI stream failed. Reason com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command REQUEST_MEASUREMENT_START error: ERROR_INVALID_STATE`

Can you please help me to understand why?

When i also tried to enable the SDK on the phone itself and at the app afterwards, i got the following error:
`SDK mode enable failed: com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command REQUEST_MEASUREMENT_START error: ERROR_NOT_SUPPORTED`

I don't understand what's the issue and would be happy for your support as soon as possible.

closed 2024-03-17 by TinDogDev

lsl-88, 2024-04-09:
@TinDogDev - Could you please share your solutions with me? I am also using Polar Ignite 3 and the application that my team developed built using the SDK was unable to pull the PPI data.

orestesgaolin, 2024-06-25:
From what I understand using the Ignite 3, user needs to enable activity mode to get HR data. Not sure if this applies to PPI, but probably that's the case.
Maybe alternatively it's possible to get the low level data when SDK mode is enabled?