polarofficial/polar-ble-sdk #449: Ignite3 integration with the BLE SDK - Getting the PPI

ozbentuneme, 2024-03-17:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Other
- [ ] Device is not relevant for this question

**Description:**
Hi,
As maintained at the Readme file, the Ignite 3 has a support for the BLE-SDK.
It supposed to provide PPI, HR and ACC.
When I used the sdk, and checked it using the getAvailableOnlineStreamDataTypes call, it is indeed supports the needed features.
The Issue is that i fail to get the PPI from the SDK.
I get the following message:
PPI stream failed. Reason com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command REQUEST_MEASUREMENT_START error: ERROR_INVALID_STATE

Can you please help me to understand why?

When i also tried to enable the SDK on the phone itself and at the app afterwards, i got the following error:
SDK mode enable failed: com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command REQUEST_MEASUREMENT_START error: ERROR_NOT_SUPPORTED

I don't understand what's the issue and would be happy for your support as soon as possible.

peterek0000, 2024-11-22:
Hi @ozbentune 

Did you manage to solve the issue?
I have the same error with ACC measurement on Polar Pacer Pro.

jimmyzumthurm, 2024-12-05:
Hi @ozbentune  and @peterek0000 ,

I apologize for late support on this matter. I added a new documentation that should you help use the SDK with Ignite 3 : https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/UsingSDKWithWatches.md Please tell me if you are to get PPI data after following these steps.

ozbentune, 2024-12-08:
Too late my friend .
Thank you

closed 2024-12-08 by ozbentune