polarofficial/polar-ble-sdk #453: one of my h10 units will not start recording.

smyffanon, 2024-03-27:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
The attached image has been ongoing about a week now. 
I've done the remove battery, hold the snaps for 30 seconds around 10 times so far.

It works fine with Polar Sensor Logger to fetch HR and ECG over bluetooth, just the memory recording refuses to start.

Any suggestions appreciated.

Could a forced firmware reload help, and if yes, how do I do that?

[image]

pth-pe-gh, 2025-02-17:
Please reported this bug to the Polar Sensor Logger app developer, since it looks like this issue appears in that app. Also, if you have problems with individual H10 sensor, please contact Polar Care for possible hardware issue resolution.

andYgnite, 2025-07-14:
@smyffanon It seems there is already a recorded file in your sensor. Did you try to delete it?
Otherwise, you won't be able to start a new one.

You can try using Android test application: https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/polar-sensor-data-collector