polarofficial/polar-ble-sdk #455: Has Anyone Successfully Achieved Real-Time Visualization of Breaths with Polar H10?

gd1925, 2024-04-07:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [/] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [/] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hello everyone,

I've been experimenting with the Polar H10 to monitor breathing patterns and I'm curious if anyone has successfully managed to get real-time visualizations of breaths.

My current setup involves post-processing the data of about 2 min with high-pass and low-pass filtfilt butterworth filters, which provides a decent representation of the signals when analyzing data for specific durations. However, when attempting to visualize the data in real-time, I face issues. The accelerometer from the Polar device operates at 200Hz, I'm sending 200 values per second for processing. So, I notice discontinuities at the end of each 200-samples, which affects the continuity and virtualization looks nothing like a breathing signal.

Has anyone faced a similar challenge or managed to overcome it? If anyone can share any strategies or adjustments I have to make to handle these discontinuities effectively. Any insights on filter adjustments, data handling techniques, or even alternative approaches to achieving smooth, real-time visualization would be greatly appreciated.

Thank you in advance for sharing your experiences and solutions.

orestesgaolin, 2024-04-18:
>  So, I notice discontinuities at the end of each 200-samples, which affects the continuity and virtualization looks nothing like a breathing signal.

We use real time PPI for display in some experiments, but due to the fact that the updates are batched (typically in 5 samples), we need to have about 2-3 seconds delay on the UI to make it look nice.

gd1925, 2024-04-28:
@orestesgaolin Ohh okay I see. Thank you for your reply.