polarofficial/polar-ble-sdk #465: How to get Timestamp with heart rate value

pranav243, 2024-06-06:
***Platform your question concerns:*** 
- [*] Android 

***Device:***
- [*] Polar H10

**Description:**
There is a timeStamp attribute with ecg and ACC values. But there isn't for HR. How to get timeStamp for the HR samples?

orestesgaolin, 2024-06-11:
I think you cannot get timestamp of the HR sample received via `polarBleApi.startHrStreaming`. On Android we just assume it's a sample from a given second `Instant.now().truncatedTo(ChronoUnit.SECONDS)`.

When it comes to offline recordings you can determine HR samples timestamps from the recording start and sampling rate:

```kt
private fun mapRecordingToBatch(recording: PolarRecording): HeartRateBatch? {
    val data = recording.data

    if (data is PolarOfflineRecordingData.HrOfflineRecording) {
        val samples = data.data.samples
        val numberOfSamples = samples.size

        /// Get the HR interval from the recording settings.
        val intervalInMs =
                recording.settings.settings[PolarSensorSetting.SettingType.SAMPLE_RATE]!!.first() * 1000

        /// Get the date of the first sample.
        val firstSampleDateUTC = data.startTime.timeInMillis + intervalInMs

        Logger.info(
                TAG,
                "Received recording data. Number of samples: $numberOfSamples. Started at: $firstSampleDateUTC"
        )

        /// Assigns a date for each hr sample. To better understand how it works, consider
        /// the following example.
        ///
        /// Let's assume that:
        /// - the first sample's date in UTC is 1670411716852 (Wed Dec 07 2022 11:15:16)
        /// - the number of samples in the exercise is 5
        /// - the interval is 1s.
        ///
        /// The consecutive samples' date such be set to:
        /// 1. 1670411716852 // 1670411716852 + 1000 * 0 (Wed Dec 07 2022 11:15:16)
        /// 2. 1670411717852 // 1670411716852 + 1000 * 1 (Wed Dec 07 2022 11:15:17)
        /// 3. 1670411718852 // 1670411716852 + 1000 * 2 (Wed Dec 07 2022 11:15:18)
        /// 4. 1670411719852 // 1670411716852 + 1000 * 3 (Wed Dec 07 2022 11:15:19)
        /// 5. 1670411720852 // 1670411716852 + 1000 * 4 (Wed Dec 07 2022 11:15:20)
        val mappedSamples = samples.mapIndexed { index, hr ->
            /// Calculate the date for each sample based on the first sample's date, interval
            /// and positional distance from the first sample.
            val date = Instant.ofEpochMilli(firstSampleDateUTC + intervalInMs * index)

            HeartRate(hr.hr, date, recording.deviceId)
        }

        val heartRateBatch = HeartRateBatch(
                samples = mappedSamples,
                recording = recording,
                deviceId = recording.deviceId
        )

        return heartRateBatch;
    }
    return null;
}
```

viesti, 2025-03-28:
Hi! I was wondering how to do that myself too, nice to see an example :)

In the example above, where does `recording.settings.settings` come from @orestesgaolin ? It seems that asking `requestOfflineRecordingSettings` for `hr` gives an error, and reading code, I think the sampling rate for HR is a static configuration, see https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift#L1210-L1211

```
        case .ppi, .hr:
            return Single.error(PolarErrors.operationNotSupported)
```

@palmqvisti can I ask from you what is the HR sampling rate of the 360 device, just to be sure :) Thanks!

palmqvisti, 2025-03-28:
@viesti, HR sample rate is 1 per second.