polarofficial/polar-ble-sdk #471: UUIDs for the Polar H10

TobiL-commits, 2024-07-07:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Dear Polar H10 Team,
at the moment, I want to read out data from my Polar H10 sensor using a python script and the bleak library. I need a UUID for my H10 device, to read the relevant sensor information and I can only find this link: https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf which is supposed to provide me this information. The document is not availiable anymore unfortunately. There would be also the possibility to read out the UUIDs directly from the sensor? Although I would not know how to go about it. 
If there is a seperate documentation to help me read data via python on my pc, or the issue needs to be handeled differently on the plattform, I would be happy to know or file another question.

Greetings 
Tobias

closed 2024-07-08 by TobiL-commits

lks-dev, 2024-08-28:
Hi Tobias,
I am in search for the same information, but sadly all links i am currently stumbling on are broken. So if you got an solution/answer i would appreciate it, if you would share your insights.
thanks.
best regards luke

Victorivus, 2024-10-29:
You have everything under: https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/

TobiL-commits, 2024-10-29:
I'm very sorry for the late answer. I missed your question. I found this document 

https://github.com/polarofficial/polar-ble-sdk/blob/8f6e9b6f8a19c6884e3eae6a5f4062f42998d879/technical_documentation/Polar_Measurement_Data_Specification.pdf

You can access the old documentation by viewing the repository history (button at the top right of the repositories path). There are also some flowcharts explaining how the data is read and so on buried somewhere, but I cannot find it again. You should be able to find it after sifting a bit through the files a bit. Hopefully you will able to progress. The new documents I didn't view yet.

Greetings 
Tobias

lks-dev, 2024-10-29:
Hi thank you two for the answers. 
In the meantime, we have stumbled upon the old commit referenced by @TobiL-commits, which was very helpful indeed.