polarofficial/polar-ble-sdk #473: Document advertisement data for each sensor to allow third party apps to use Apple AccessorySetupKit in iOS 18

edorphy, 2024-07-09:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [x] Polar Verity Sense
- [x] Polar H10
- [x] Polar H9
- [x] Other
- [ ] Device is not relevant for this question

**Description:**
Apple introduced a new framework for 'associating' Bluetooth sensors to an app coming out this fall in iOS 18. I got it to work finally with my Polar devices that I own, but I can't confirm it will work with your other devices without some info.

In order enable Bluetooth enabled fitness apps to use your sensors with the new AccessorySetupKit, please document the following:
- [ ] Local Name of each device in the advertisement payload
- [ ] Advertisement Data payload format, as much as possible. Looking specifically for Mfg Data record
- [ ] What official images should developers use for each of your products to present to our users

Please find the documentation here: 
https://developer.apple.com/documentation/accessorysetupkit/discovering-and-configuring-accessories

The accompanying WWDC24 video here:
https://developer.apple.com/wwdc24/10203

And finally the sample code here: 
https://developer.apple.com/documentation/accessorysetupkit/authorizing-a-bluetooth-accessory-to-share-a-dice-roll

For demonstration purposes, I took a screenshot of your product from a product page to get this code to show your device in the new picker UI:

```swift
    private static let polarVeritySense: ASPickerDisplayItem = {
        let descriptor = ASDiscoveryDescriptor()
        descriptor.bluetoothServiceUUID = CBUUID(string: "180D")
        descriptor.bluetoothNameSubstring = "Polar Sense"

        return ASPickerDisplayItem(name: "Polar Sense", productImage: UIImage(named: "PolarVeritySense")!, descriptor: descriptor)
    }()

    private static let polarH10: ASPickerDisplayItem = {
        let descriptor = ASDiscoveryDescriptor()
        descriptor.bluetoothServiceUUID = CBUUID(string: "180D")
        descriptor.bluetoothNameSubstring = "Polar H10"

        return ASPickerDisplayItem(name: "Polar H10", productImage: UIImage(named: "PolarH10")!, descriptor: descriptor)
    }()
```

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>NSAccessorySetupBluetoothServices</key>
	<array>
		<string>180D</string>
	</array>
	<key>NSAccessorySetupKitSupports</key>
	<array>
		<string>Bluetooth</string>
	</array>
	<key>NSAccessorySetupBluetoothCompanyIdentifiers</key>
	<array>
		<string>006B</string>
	</array>
</dict>
</plist>

```

[image]

OmniPrecedence, 2024-07-14:
@edorphy Thank you so much, can you possibly provide me the exact steps to implement real time heart data to my personal app using 'Polar Ignite 3'. If have a general guideline that would help, I would mostly likely be happy as Im tight on schedule and I want to be able to see my heart rate data on my custom application on iPhone and be able to send it to my contacts using Twilio. I'm asking because you seem to know  what your doing and worked with iOS and SDK before, Thank you.

edorphy, 2024-07-14:
> @edorphy Thank you so much, can you possibly provide me the exact steps to implement real time heart data to my personal app using 'Polar Ignite 3'. If have a general guideline that would help, I [...]

Are looking for guidance specifically on how to associate the Polar Ignite 3 with your app using the new AppleAccessoryKit framework available in iOS 18 beta and publicly available in the fall? Or are you just looking for information to connect your Ignite product to any iOS app today as-is?

If the first, I can't help because I don't own one to look at its advertisement data to know what to supply to Apple for the new framework. That is exactly what this GitHub issue was asking for. Presumably you'd be able to use my sample code above, but that makes unconfirmed assumptions about the advertisement of the Ignite device.

If the second, according to this SDK readme, Ignite 3 is supported. You should be able to scan, discover, and connect to your sensor all directly using this framework. I would check out the documentation Polar provides and their example project. 

Regardless of which you're looking for help with, I would highly recommend that you carefully review Apple's "App Review Guidelines" against your use case. You should be fine so long as the user understands your app will transmit the health data off device; informing your users of this behavior in in your app UI experience and your privacy statement is a good start. It doesn't sound like you're using HealthKit which would have more rules and instead you want to get heart rate directly from the sensor. Even so, the best guidance I can share is read the App Review Guidelines.

OmniPrecedence, 2024-07-15:
No I just need the example from the sdk to work for my project, and i wanted to know what pieces of code you grabbed for the sdk to work, i have polar ignite 3 and im building an iOS app