polarofficial/polar-ble-sdk #477: Current version of Polar SDK cannot compile for iOS

orestesgaolin, 2024-07-24:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this bug

**Describe the bug**

Trying to run current master on iOS fails with following exception

> Swift Compiler Error (Xcode): The compiler is unable to type-check this expression in reasonable time; try breaking up the expression into distinct sub-expressions
> .../ios/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift:1266:19

[image]

**How to Reproduce**

Add PolarBleSDK as a dependency to the iOS project

```
  pod 'PolarBleSdk', :git => 'git@github.com:polarofficial/polar-ble-sdk.git', :branch => 'master', :commit => 'e436085dc09a3d8a07a5c525174fb24875d3b613'
```

**Expected behavior**

Should compile same as 5.5.0

ashayk, 2024-07-26:
Confirmed, I'm having the same issue.

samulimaa, 2024-07-31:
Hello, please try again with the latest master branch revision 5b7239fa018f88cf0ab61a4299ca04f3b3875267.

closed 2024-08-30 by orestesgaolin