polarofficial/polar-ble-sdk #478: Signal LED blinking cannot be turned off during PPI streaming on Polar Verity Sense and Polar OH1

petrnalevka, 2024-07-25:
**Platform on which you observed the bug:** 
- [X] Android 
- [X] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [X] Polar OH1
- [X] Polar Verity Sense

**Describe the bug**

When PPI streaming starts user facing LED starts blinking in purple color. Even when I use:
PolarBleApi.PolarBleSdkFeature.FEATURE_POLAR_LED_ANIMATION
and then calling
api.setLedConfig(device.deviceId, LedConfig(false, false))
even when this is called after the PPI stream is established

**How to Reproduce**

api.startPpiStreaming(device.deviceId)
and then call
api.setLedConfig(device.deviceId, LedConfig(false, false))

**Expected behavior**

I expect the LED be turned off when I call api.setLedConfig(device.deviceId, LedConfig(false, false))

Many thanks for looking into this..

petrnalevka, 2024-07-28:
Hello @jimmyzumthurm  big thanks for getting back to me.. 

I did update firmware on my Polar Verity Sense to 2.2.6..

[image]

I'm calling the setLedConfig(device.deviceId, LedConfig(false, false)) after the LED_ANIMATION feature is ready and then again after the PIP streaming is established, but still the purle bliking led is on.. 

Any tips what I could try? 

Many thanks for your help!

jimmyzumthurm, 2024-07-31:
@petrnalevka That is quite surprising. On my side this feature seems to work well... If I enable PPI streaming, then disable the LED it turns it off immediately and re-enabling turns it on immediately as well, I'm not able to reproduce this. 

@samulimaa Any ideas?

petrnalevka, 2024-07-31:
@jimmyzumthurm many thanks for giving it try, really strange I will do more tests on my end..

petrnalevka, 2025-10-08:
Hello @palmqvisti many thanks I'm still seeing the issue. I have updated firmware on both of my device Verity Sense on OH1+ and still teh led light is blinking even being disabled.. I will check out the new examples.. many thanks..