polarofficial/polar-ble-sdk #480: Cannot list offline recordings on Polar 360

vedantgurav, 2024-08-12:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other `Polar 360`
- [ ] Device is not relevant for this bug

**Describe the bug**
In the iOS SDK demo project, I can successfully start and stop offline recordings on the Polar 360. But when I try to list the recordings, I get operationNotSupported.

Additionally, the disk space remains unchanged on the device.

**How to Reproduce**
- Create an offline HR recording on Polar 360
- Navigate to Offline tab > Read tab

**Expected behavior**
- Get a list of offline recordings

**Screenshots and logs**
```
Start offline recording listing
2024-08-07 11:26:38.770: listOfflineRecordings -> subscribed
2024-08-07 11:26:38.782: listOfflineRecordings -> Event error(operationNotSupported)
Offline recording listing error: operationNotSupported
2024-08-07 11:26:38.782: listOfflineRecordings -> isDisposed
```
|List Offline Empty|Unchanged Disk Space|
|-|-|
|[image]|[image]|

hamletAppsfactory, 2024-09-23:
Issues i could reproduce with **Polar 360** device (**Firmware version: 1.1.0**) :

- Offline recording was inactive for all the type of measurements.
- Online recording didn't work. You can see the log attached below.
- I couldn't get the time from the device.

[image]
[image]
[image]
[image]

vedantgurav, 2024-09-30:
Solved by fixing the FileSystemType function. Polar 360 was missing here. https://github.com/polarofficial/polar-ble-sdk/blob/f9c57d5148de25199544338a4a3a65125792da4a/sources/iOS/ios-communications/Sources/iOSCommunications/ble/api/model/polar/BlePolarDeviceCapabilitiesUtility.swift#L21-L39

closed 2024-09-30 by vedantgurav