polarofficial/polar-ble-sdk #481: Example iOS app raises "Cannot find type 'Communications_PbPFtpOperation' in scope"

chrispreee, 2024-08-14:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ x] Device is not relevant for this bug

**Describe the bug**
I cloned the Polar SDK repo, ran `pod install` in the root of the example iOS project, opened the project in XCode and clicked build, and the build fails with: "Cannot find type 'Communications_PbPFtpOperation' in scope"

**How to Reproduce**
I have done this from a fresh clone.

Prior, I have run 

```
brew install swift-protobuf
sudo gem install cocoapods
pod setup
pod repo update
```

Running XCode 15.4.

**Expected behavior**
I expect the project to find the Protobuf dependency and build successfully.

**Screenshots and logs**

Full build log:

PolarBleSdk
...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/protobuf/pftp_request.pb.swift
...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/protobuf/pftp_request.pb.swift:190:11 Cannot find type 'Communications_PbPFtpOperation' in scope

...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift
...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift:1458:94 'PolarOhrData' is deprecated: renamed to 'PolarPpgData'

...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift:1873:33 'PolarOhrData' is deprecated: renamed to 'PolarPpgData'

RxSwift
...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/Pods.xcodeproj The iOS Simulator deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to 9.0, but the range of supported deployment target versions is 12.0 to 17.5.99.

SwiftProtobuf
...polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/Pods.xcodeproj The iOS Simulator deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to 9.0, but the range of supported deployment target versions is 12.0 to 17.5.99.

ymatt345, 2024-09-09:
I am having the same issue with PolarBleSdk 5.7.0.
I hope this gets fixed soon.

ymatt345, 2024-09-09:
PolarBleSdk 5.7.0 in Pods does not have Communications_PbPFtpOperation.pb.
However, polarBleSdk 5.4.0 in Pods has Communications_PbPFtpOperation.pb.
So Communications_PbPFtpOperation.pb might be missing from polarBleSdk 5.5.0 and later.
How can we include Communications_PbPFtpOperation.pb in polarBleSdk in Pods?
[image]
[image]

chrispreee, 2024-09-09:
I copied in the file to:

`polar-ble-sdk/examples/example-ios/iosBleSdkTestApp/Pods/PolarBleSdk/sources/iOS/ios-communications/Sources/iOSCommunications/ble/api/model/protobuf/communications_pftp_request.pb.swift`

Then in XCode right-click on the `PolarBleSdk` `Pod`, then add the file to the project, then it seems to build, and successfully install and record.

I haven't created a PR because someone more experienced than I should confirm that this is all that needs doing or all that is missing.

ymatt345, 2024-09-09:
Okay, thank you. I'll try to fix it.
However, the bleSDK should be fixed by their developer.

chrispreee, 2024-09-09:
If you have problems let me know and I'll push my project, but I'm very me to iOS development.

hamletAppsfactory, 2024-10-22:
I also have the same issue, is there any solution yet?

kasperdijkstra, 2024-11-15:
I can confirm the proposed solution by Chrispreee works. I opened a pull request.

samulimaa, 2024-11-19:
Great job fixing this, PR merged.