polarofficial/polar-ble-sdk #484: Timestamp for PPI measurements.

hamletAppsfactory, 2024-08-27:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this question

Why the Timestamp for PPI (PolarPpiData) measurements are deprecated and always 0? Is it possible to get them somehow?

orestesgaolin, 2024-08-30:
In our case we just assume that the last PPI sample comes from current instant, and samples before that are x ms before that e.g. if we get a batch of `[300,500,900]` ms then they are at `[t-(300+500+900), t-(500+900), t-900]` where t is `now`

hamletAppsfactory, 2024-09-10:
Thanks for the reply, let's assume this is not an offline recording and there is no startTime provided by the device(or SDK).
What is **_now_** in this case: 
1) the time of receiving the first measurement
2) the time of sending the command to the device to start the streaming?
In my case with OH1 there is a difference between those two points around 3-5 seconds.

jimmyzumthurm, 2024-10-29:
@orestesgaolin is correct. I think it's even 25 seconds to receive the first sample after the start command, which contains 5 second windows of PPI samples. The algorithm inside Verity Sense doesn't calculate any timestamp for the samples and this is why it's not provided for PPI data for Verity Sense. 

Some other devices such as Polar 360 do have timestamps attached to PPI data as they have more recent algorithm that provide it.

closed 2024-10-29 by jimmyzumthurm