polarofficial/polar-ble-sdk #491: `listOfflineRecordings` doesn't list all recordings

metuuu, 2024-10-23:
**Platform on which you observed the bug:**
- [X] Android (polar-ble-sdk 5.7.0)
- [?] iOS (not tested)

**Device on which you observed the bug:**
- [X] Polar Verity Sense (Software: 2.2.6)

**Describe the bug**
Listing sensor measurements might return only single item even though there are multiple recordings. After the first recording is removed from the sensor, then polar ble sdk returns the other item.
I have not find a reason for this or a way to reproduce this consistently.

I had this issue happening 2/3 of the time but in the next day it stopped and never happened for me again. After I released our app to testing, our tester had the same issue where he described that approximately every 1/3 recording is missing ACC. And after the recording is removed from sensor then the ACC appears. The recordings were approx 10min - 1h long. (When I had the issue, the recording time didn't affect the outcome. I tested 10sec - 10min recordings)

**How to Reproduce**
1. Start PPI+ACC or PPG+ACC recordings with Verity Sense.
2. Wait a moment and stop both recordings
3. List recordings -> only one of the two or both recordings are returned
4. If only one of the two recording was returned, delete the returned recording (If both were returned, repeat until only single one is returned)
5. List recordings -> now the missing recording should appear

**Expected behavior**
All recordings should be returned when listing offline recordings

**The code used in our app for listing the offline recordings**
```kt
    @SuppressLint("CheckResult")
    @ReactMethod
    fun listOfflineRecordings(callback: Callback) {
        if (deviceId == null) return callback.invoke("No device connected")
        api!!.listOfflineRecordings(deviceId!!)
            .observeOn(AndroidSchedulers.mainThread())
            .doOnSubscribe { offlineRecordingEntries = mutableListOf() }
            .subscribe(
                { offlineRecordingEntries!!.add(it) },
                { error ->
                    if (error is PolarOperationNotSupported) {
                        callback.invoke(ERROR_FEATURE_NOT_SUPPORTED, deviceId)
                    } else {
                        Log.d(TAG, "list offline recording failed: " + throwableToString(error))
                        callback.invoke(throwableToString(error), deviceId)
                    }
                },
                {
                    val result = Arguments.createMap()
                    val entriesArray = Arguments.createArray()
                    offlineRecordingEntries!!.forEach { entry ->
                        val entryMap = Arguments.createMap()
                        entryMap.putString("type", entry!!.type.name.lowercase())
                        entryMap.putDouble("size", entry.size.toDouble())
                        entriesArray.pushMap(entryMap)
                    }
                    result.putArray("entries", entriesArray)
                    callback.invoke(null, result)
                }
            )
    }
```

samulimaa, 2024-11-01:
Hi, thanks for reporting this, we have created internal ticket to fix this on upcoming releases.

metuuu, 2024-11-01:
Thank you! Were you able to reproduce the issue?

samulimaa, 2024-11-01:
We are working on it.

closed 2025-03-21 by palmqvisti