polarofficial/polar-ble-sdk #497: Import Pole BLE on Meta Quest 3

PaoloSalvatico, 2024-11-04:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I have sucessfully build an apk to use on Meta Quest 3, i used Unity 6 editor cause i needed a version of android NDK high enough to support polar sdk, the probles is that after i open the apk file on quest3 the InitializePolarBleSdk()() function returns the following error: "Error initializing Polar BLE SDK: java.lang.ClassNotFoundException;com.polar.sdk.api.PolarBleApiDefaultImpl Polar BLE Manager is only supported on Android devices."

**How to Reproduce**
This is the function incriminted i took from the demo in polar sdk:

void InitializePolarBleSdk()
    {
        try
        {
            // Get the current Android Activity
            using (AndroidJavaClass unityPlayer = new AndroidJavaClass("com.unity3d.player.UnityPlayer"))
            {
                context = unityPlayer.GetStatic<AndroidJavaObject>("currentActivity");
            }

            // Create an instance of PolarBleApi
            using (AndroidJavaClass polarBleApiClass = new AndroidJavaClass("com.polar.sdk.api.PolarBleApiDefaultImpl"))
            {
                // Set the features you want to use
                var features = new AndroidJavaObject("java.util.HashSet");
                features.Call<bool>("add", "FEATURE_HR");

                // Get default implementation
                polarBleSdk = polarBleApiClass.CallStatic<AndroidJavaObject>("defaultImplementation", context, features);
            }

            // Set API callback
            SetApiCallback();
        }
        catch (Exception ex)
        {
            UpdateStatus("Error initializing Polar BLE SDK: " + ex.Message);
        }
    }

**Expected behavior**
Meta should be an Android device, my question is: Did i miss something or polr sdk doesn't work with Quest 3?
Thanks

palmqvisti, 2025-10-07:
Hi @PaoloSalvatico, are you still working on this issue? Did you try our latest SDK release and did it fix the issue?

PaoloSalvatico, 2025-10-08:
Hello @palmqvisti, I did resolve the issue and successfully connected polar BLE to Meta quest 3 device! Thanks anyway

closed 2025-10-08 by palmqvisti