polarofficial/polar-ble-sdk #505: Infinite loop when getting sleep for a single day

orestesgaolin, 2024-11-20:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this bug

**Describe the bug**

When fetching sleep data via `polarBleApi.getSleep(deviceId, fromDate, toDate)` if you provide `fromDate` equal to `toDate` it ends up with an infinite loop because `getDatesBetween` function uses `while (startDate == endDate || endDate.isAfter(theDate))`.

[See the implementation](https://github.com/polarofficial/polar-ble-sdk/blob/85f58fdf6e74641b3dd8ddda7c423e6c6ad9ecbc/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/BDBleApiImpl.kt#L2112-L2123)

**How to Reproduce**

Call `polarBleApi.getSleep(deviceId, fromDate, toDate)` with `fromDate = toDate`.

**Expected behavior**

SDK should return sleep for the given day.
Imho it's enough to update `while` condition to `while (theDate == endDate || endDate.isAfter(theDate))`. We may be able to contribute PR with a fix soon.

**Screenshots and logs**

pth-pe-gh, 2024-11-26:
FLOW-60719

samulimaa, 2024-12-13:
This should be fixed in the latest SDK release 5.10.0.

orestesgaolin, 2024-12-19:
Before
```kt
    private fun getDatesBetween(startDate: LocalDate, endDate: LocalDate): MutableList<LocalDate> {
        var theDate: LocalDate = startDate
        var datesList = mutableListOf<LocalDate>()

        while (startDate == endDate || endDate.isAfter(theDate)) {
            datesList.add(theDate)
            theDate = theDate.plusDays(1)
        }

        return datesList

    }
```

After

```kt
    private fun getDatesBetween(startDate: LocalDate, endDate: LocalDate): MutableList<LocalDate> {
        var theDate: LocalDate = startDate
        var datesList = mutableListOf<LocalDate>()

        while (theDate == endDate || endDate.isAfter(theDate)) {
            datesList.add(theDate)
            theDate = theDate.plusDays(1)
        }

        return datesList
    }
```

closed 2024-12-19 by orestesgaolin