polarofficial/polar-ble-sdk #509: Crash in `parsePmdSettingsData`

vedantgurav, 2024-11-28:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other (Polar 360)
- [ ] Device is not relevant for this bug

**Describe the bug**

Occasionally seeing a crash here when getting accelerometer settings for online stream

https://github.com/polarofficial/polar-ble-sdk/blob/734c1f1169cda6ea8aa438e6e02faf5e636f11a3/sources/iOS/ios-communications/Sources/iOSCommunications/ble/api/model/gatt/client/pmd/PmdSetting.swift#L47

**Screenshots and logs**
[image]
[image]

rkangast, 2024-11-29:
FLOW-60819

pth-pe-gh, 2025-02-12:
Hi @vedantgurav - are you still experiencing this with latest SDK? Can you provide any more details on this crash, e.g. request and related parameters to start online measurement you are using when experiencing this?

vedantgurav, 2025-02-12:
Hi @pth-pe-gh 

This crash occurred when I try to get the accelerometer online stream settings.

However, I only fetch these settings to check if it throws the device in charger error. I do this because there isn't a way to check for charging state yet. I am not trying to actually start an online stream.

I wasn't able to reproduce it myself, but I saw a few crash reports from this function. Finally since I don't actually need the settings dictionary, I have modified this function to return an empty dictionary to avoid the crash in production. So cant say if it is still a problem.

pth-pe-gh, 2025-02-12:
Cannot reproduce using recent SDK code.  I will close this bug for now as there is a workaround in use. It can be re-opened if it can be reproduced with latest SDK version, and latest device firmware version.

closed 2025-02-12 by pth-pe-gh

pth-pe-gh, 2025-02-12:
@vedantgurav - in addition to the version infos, raw response bytes would help if you ever capture them from this crash.