polarofficial/polar-ble-sdk #510: Crash in `CBDeviceListenerImpl.search`

vedantgurav, 2024-11-28:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other (Polar 360)
- [ ] Device is not relevant for this bug

**Describe the bug**
Encountered a crash in `CBDeviceListenerImpl.search` at the `CBScanner.addClient` call when using the `searchForDevice` SDK function.

https://github.com/polarofficial/polar-ble-sdk/blob/734c1f1169cda6ea8aa438e6e02faf5e636f11a3/sources/iOS/ios-communications/Sources/iOSCommunications/ble/endpoints/corebluetooth/central/CBDeviceListenerImpl.swift#L281-L288

**Screenshots and logs**
[image]
[image]

rkangast, 2024-11-29:
FLOW-60821

pth-pe-gh, 2025-01-02:
@vedantgurav - do you encounter this same crash using current iOS example app code? If not, is it possible for you to supply code sample that reproduces the crash?

closed 2025-10-07 by palmqvisti