polarofficial/polar-ble-sdk #526: Do FirstTimeUse on Polar 360

ymatt345, 2024-12-11:
***Platform your question concerns:*** 
- [ ] Android 
- [✓] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [✓] Polar 360
- [ ] Device is not relevant for this question

I'm trying to add support for Polar 360 in an app that supports other Polar sensor devices.
I'm starting with BLE SDK v5.9.0 and its sample app.
The sample app doesn't have a doFirstTimeUse() function.
The FirstTimeUse.md document doesn't list the command and parameters to use doFirstTimeUse().
How do I run doFirstTimeUse()?
Please let me know if there are any special procedures I need to follow.

hamletAppsfactory, 2025-01-07:
Example:

```
api.doFirstTimeUse(
                deviceId,
                ftuConfig:
                    PolarFirstTimeUseConfig(
                        gender: .female,
                        birthDate: .now,
                        height: 170.0,
                        weight: 50.0,
                        maxHeartRate: 120,
                        vo2Max: 70,
                        restingHeartRate: 90,
                        trainingBackground: .regular,
                        deviceTime: Date().dateAndTimeISO8601(),
                        typicalDay: .mostlySitting,
                        sleepGoalMinutes: 420
))
```

hamletAppsfactory, 2025-01-07:
Is there a way to check whether doFirstTimeUse() was called for the device ?

ymatt345, 2025-01-11:
I could find the api.doFirstTimeUse in Polar-ble-sdk v5.11.0.
So I will test it soon.
Thank you.

gaurav36, 2025-05-26:
I have same issue while connecting this device via BLE GATT ? 

```bash
{
  "technology": "ble",
  "ids": ["DEVICE_ID"],
  "controlApp": "controlAppName",
  "ble": {
    "type": "gatt",
    "serviceID": "fb005c80-02e7-f387-1cad-8acd2d8df0c8",
    "characteristicID": "fb005c81-02e7-f387-1cad-8acd2d8df0c8",
    "value": "BASE64_ENCODED_COMMAND_BYTES",
    "operation": "write"
  }
}
```

and if serviceID "fb005c80-02e7-f387-1cad-8acd2d8df0c8" and characteristicID "fb005c81-02e7-f387-1cad-8acd2d8df0c8" is correct ?

Can you tell me what value we need to use here?

rkangast, 2025-07-04:
> Is there a way to check whether doFirstTimeUse() was called for the device ?

Hi @hamletAppsfactory 

Sorry for a very late response. You can use this [API method](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#isFtuDone(java.lang.String)) to see if the FTU has already been done to the device.

closed 2025-07-04 by rkangast