polarofficial/polar-ble-sdk #529: RX Java Exception - UndeliverableException

talbigelbinah, 2024-12-15:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
We had a crash/error report from Sentry:

ProtocolViolationException
 
io.reactivex.rxjava3.exceptions.UndeliverableException in <init>
```
com.polar.androidcommunications.api.ble.exceptions.BleAttributeError: Failed to set characteristic notification or indication  failed with error: 133
    at com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1
    at com.google.firebase.messaging.FcmBroadcastProcessor$$ExternalSyntheticLambda1.subscribe(FcmBroadcastProcessor:0)
    at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate.subscribeActual(CompletableCreate.java:40)
    at io.reactivex.rxjava3.core.Completable.subscribe(Completable.java:2860)
    at io.reactivex.rxjava3.internal.operators.completable.CompletableSubscribeOn$SubscribeOnObserver.run(CompletableSubscribeOn.java:64)
    at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
    at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
    at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
    at java.util.concurrent.FutureTask.run(FutureTask.java:264)
    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
    at java.lang.Thread.run(Thread.java:1012)
io.reactivex.rxjava3.exceptions.ProtocolViolationException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | com.biosensesignal.sdk.zn: Failed to set characteristic notification or indication  failed with error: 133
    at io.reactivex.rxjava3.exceptions.UndeliverableException.<init>
    at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:372)
    at io.reactivex.rxjava3.internal.util.AtomicThrowable.tryAddThrowableOrReport(AtomicThrowable.java:52)
    at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError
    at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError(FlowableFlatMap.java:636)
    at io.reactivex.rxjava3.internal.operators.flowable.FlowableFromCompletable$FromCompletableObserver.onError(FlowableFromCompletable.java:83)
    at io.reactivex.rxjava3.internal.operators.completable.CompletablePeek$CompletableObserverImplementation.onError(CompletablePeek.java:95)
    at io.reactivex.rxjava3.internal.operators.completable.CompletableObserveOn$ObserveOnCompletableObserver.run(CompletableObserveOn.java:87)
    at io.reactivex.rxjava3.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:123)
    at android.os.Handler.handleCallback(Handler.java:959)
    at android.os.Handler.dispatchMessage(Handler.java:100)
    at android.os.Looper.loopOnce(Looper.java:232)
    at android.os.Looper.loop(Looper.java:317)
    at android.app.ActivityThread.main(ActivityThread.java:8705)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
```

**How to Reproduce**
We don't know how to reproduce. This is a sentry issue we received. It looks like something in the Polar lib

TeemuLauPolar, 2024-12-16:
FLOW-61107

orestesgaolin, 2024-12-19:
To avoid crashes in the meantime you can use following code ([see more here](https://medium.com/@s.vinouze/default-error-handling-with-rxjava-89b79de3f125)):

```kt
RxJavaPlugins.setErrorHandler {
  Logger.error(TAG, "Error from RxJavaPlugins.setErrorHandler", it)
  // send error to Sentry etc.
}
```

palmqvisti, 2025-10-07:
The fix for this issue has been released and is included in recent versions of SDK. I'm closing this issue due to inactivity. If you still face problems, please report a new issue.

closed 2025-10-07 by palmqvisti