polarofficial/polar-ble-sdk #533: Automatic update dependencys

BDVGitHub, 2024-12-29:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this question

**Description:**
Is it possible to use dependabot to keep dependencys up to date https://docs.github.com/en/code-security/getting-started/dependabot-quickstart-guide or [Renovatebot](https://github.com/renovatebot)

pth-pe-gh, 2025-01-16:
We will discuss this suggestion with team.

pth-pe-gh, 2025-01-17:
We have plans to update dependency checking for BLE SDK code. While we do that we will check in detail if Dependabot can improve the current process of updating dependencies in timely, safe, and productive manner.  

Thanks for pointing out this possible improvement!

pth-pe-gh, 2025-01-17:
FLOW-61716