polarofficial/polar-ble-sdk #537: No such module 'Zip'

stuffmc, 2025-01-12:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this bug

**Describe the bug**
I did a blank App where I only added the Swift Package. Compiling doesn't even work.

**How to Reproduce**
See attached file, and compile.

**Expected behavior**
It should compile.

[PolarStuff.zip](https://github.com/user-attachments/files/18390067/PolarStuff.zip)

pth-pe-gh, 2025-03-19:
Verified. Current workaround is to use Cocoapods instead of Swift Package Manager.

closed 2025-03-25 by pth-pe-gh

pth-pe-gh, 2025-03-25:
Note that before next release, one needs to use branch master when integrating Polar BLE SDK using Swift Package Manager.