polarofficial/polar-ble-sdk #538: Transfer speed of offline recording to device Polar Verity Sense?

abbjetmus, 2025-01-13:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi!

I wonder what we can expect for transfer speed of offline recordings sent to the mobile app using the sdk. I'm not using Sdk Mode.
If the memory has storage limit 2MB, how long time does that take to transfer?

/Regards

jimmyzumthurm, 2025-01-13:
Hi @abbjetmus 

The storage capacity of Verity Sense is more about 14 MB with a "cleaned" memory (ex: after a factory defaults). When less than 2 MB are left, then starting new offline recordings it not possible. 

Speed depends on what kind of platform you are using. I've tested with Android, and speed is around 15 KB/s. For iOS it might be different because the BLE connection interval that is negotiated for faster speed are between 7-11ms for Android, but 15 ms for iOS so it could be a bit slower. So it will take quite a while to sync 14 MB of data at once. 

Polar 360 which is more recent device that also supports this feature has a much better transfer speed, which can reach as high as ~80 KB/s with Android.

abbjetmus, 2025-01-13:
@jimmyzumthurm I see but why is the speed so low, bluetooth standard speed should be 2MB/s with version 2.0?
Is there any work being done to increase this?

jimmyzumthurm, 2025-01-13:
@abbjetmus  2 MB/s is the maximum achieve speed of transfer that is possible over the protocol. Verity Sense is a small embedded device with constrained resources and there some overhead to get the data from the memory back to the phone. There is no work being done to increase this for this product, but you can expect newer generation of products to match the speed of Polar 360.

abbjetmus, 2025-01-13:
Thank you @jimmyzumthurm, Polar 360 might be a better option for us. What are the specs for it.
Can't find anything regarding of pricing?
What is the offline memory size and what transfer speed can we expect?

jimmyzumthurm, 2025-01-13:
Please use https://www.polar.com/en/business/contact-us to get more details about Polar 360 pricing and availability. I don't have any information yet about transfer speed with iOS.