polarofficial/polar-ble-sdk #539: Progress bar for `getOfflineRecord`

metuuu, 2025-01-15:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this question

**Description:**
Reading offline record (`getOfflineRecord`) from sensor (I am testing with Polar Verity Sense) takes a long time load from sensor (especially with longer measurements). Is it possible to track the data transfer progress percentage or would you be willing to implement it?

pth-pe-gh, 2025-01-17:
Thanks for the question. Currently this is not possible, but it should be possible to implement. We recorded the this improvement suggestion as backlog item.

pth-pe-gh, 2025-01-17:
FLOW-61713

rkangast, 2025-10-30:
New APIs for this has been added in SDK release 6.9.0. See release notes for details.