polarofficial/polar-ble-sdk #541: Device support

Larrysea, 2025-01-21:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [  ] Other
- [ x] Device is not relevant for this question

**Description:**
May I ask if these（unite, vantage m3，ignite2 ，vantage v2，grit x pro ） devices will have a chance to be supported in the future? Additionally, will newly released devices have their SDKs updated promptly to ensure support? 😊

rkangast, 2025-01-23:
FLOW-62115

tmiland-labs, 2026-10-14:
Hi! Vantage V2 support is now proposed for review in #897 — standard file system access (activity, sleep, training exports) plus device management, verified on real hardware.

One note on streaming: the V2's firmware doesn't expose the PMD service (`FB005C80`), so online streaming isn't available today — but if Polar ever ships a firmware update that exposes PMD, the SDK will pick it up automatically (feature readiness is based on service discovery at connect time, not a hard-coded device list) and streaming will work with the SDK as-is. :)